SystemRDL/systemrdl-compiler #10: Perl preprocessor is too restrictive

RasmusGOlsen, 2018-10-18:
I can't load Perl modules with "use" or "require", then I get a Perl syntax error.
```
atxmega_spi.rdl:1:0: error: Encountered a Perl syntax error while executing embedded Perl preprocessor commands:
'require' trapped by operation mask at (eval 5) line 1, <> line 8.

<% use strict; %>
```

 The system environment variables are also not accessible. Do the Perl preprocessor need to run in a Safe compartment, if not just use `eval($miniscript);` else please add possibility to load files and access environment, by adding this.

https://github.com/SystemRDL/systemrdl-compiler/blob/master/systemrdl/preprocessor/ppp_runner.pl line 35.
```perl
# Run miniscript in restricted context
my $compartment = new Safe;
$compartment->permit(':load');
$compartment->share_from('main', [
    '%ENV',
    'rdlppp_utils::emit_ref',
    'rdlppp_utils::emit_text'
]);
$compartment->reval($miniscript);
```

amykyta3, 2018-10-19:
Thanks for finding this!
I had definitely intended to provide access to `%ENV` but somehow missed that one.
Also added access to `use` and `require`, as well as read access to the filesystem, and some extra math operations.

I'll include this change in the next release

amykyta3, 2018-10-24:
Fixed in v1.2.0 release

closed 2018-10-24 by amykyta3