SystemRDL/systemrdl-compiler #11: interal/external properties don't propogate

bdmagnuson, 2018-10-31:
I've tried annotating elements with external and while this is accepted by the parser the following visitor gets `None` for all nodes.  I've tried putting `external` (or `internal`) on the component def,  an instance of an anon def, and on an explicit instance.  All the same.

```
class RTLListener(RDLListener):
    def __init__(self):
        pass

    def enter_Component(self, node):
        print(node.inst.external)
```

As an aside the parser rejects this.  Admittedly a weird thing to do, but I think it's legal.

```
   external reg {
       field {} f;
    } external r;
```

amykyta3, 2018-11-01:
Thanks for reporting!
I had this on my todo list but looks like I lost track of it. Should be pretty quick to implement internal/external propagation.

I'll double-check if the second example is valid. At a glance, I'm not sure if the grammar allows it but it is worth a closer look.

bdmagnuson, 2018-11-01:
No problem!  Happy to test.  I did check the grammar and the 'double' component_inst_type formulation is indeed not legal.  I just happened to enter that by accident when I was testing permutations and thought 'huh.  well why not?'

Your project looks great though.  I have my own at https://github.com/bdmagnuson/hsrdl that I've been meaning to

1. Clean up and make more idiomatic (this was my first Haskell project)
2. Support SRDL 2.0

Stumbling upon your project saved me a lot of time!  At least for my immediate needs, even if I do have to whip up an RTL backend for it, which is precisely how I encountered the external/internal issue.

At the very least it's been interesting to see how someone else interprets some of the fuzzier parts of the spec.  I still think that my favorite one is:

> 'A SystemRDL compiler shall imply the nature of a counter as a up counter, a down counter, or an up/down counter by the properties specified for that counter field.'

amykyta3, 2018-11-01:
If you're working on your own implementation, be sure to check out my [implementation notes](https://github.com/SystemRDL/systemrdl-compiler/tree/master/docs/implementation_notes). These are basically excerpts from my "lab notebook" that document my reasoning for some of my design choices. Some of the stuff is pretty specific to my implementation, but maybe there will be something useful for your project. I'm also maintaining a document on any [RDL spec issues/ambiguities](https://github.com/SystemRDL/systemrdl-compiler/blob/master/docs/implementation_notes/systemrdl_spec_issues.md) I've come across. These are less project-specific and may be useful.

If you're putting together an RTL back-end, I'd be interested in seeing what you've done! I'm slowly putting together some exporters such as [HTML](https://github.com/SystemRDL/RALBot-html) and [IP-XACT](https://github.com/SystemRDL/RALBot-ipxact). Eventually these components may get tied together into a coherent command-line tool.

bdmagnuson, 2018-11-05:
Seems pretty good on initial inspection.  Thanks for taking care of this so quickly!

closed 2018-11-05 by bdmagnuson