SystemRDL/systemrdl-compiler #12: Nested parameters don't reach leaf instances

bdmagnuson, 2018-11-02:
Given this code the fields should have a width of 4.  Instead they are taking the default value of the enclosing regfile (i.e. 2)

```
addrmap nested {

    reg r1 #(longint unsigned WIDTH=1)  {
        field {} f[WIDTH];
    };

    regfile rf1 #(longint unsigned WIDTH=2) {
        reg r2 #(longint unsigned WIDTH=3)  {
            field {} f[WIDTH];
        } #(.WIDTH(WIDTH)) r_inst1;

        r1 #(.WIDTH(WIDTH)) r_inst2;
    };

    rf1 #(.WIDTH(4)) rf_inst;

};
```

amykyta3, 2018-11-03:
confirmed
https://travis-ci.org/SystemRDL/systemrdl-compiler/builds/450118525

bdmagnuson, 2018-11-05:
This also looks like it's fixed with my small test case.

closed 2018-11-05 by bdmagnuson