SystemRDL/systemrdl-compiler #14: No module named 'typing'

Muffo, 2018-11-15:
I installed the library using the recommended command, that is:

```
python3 -m pip install systemrdl-compiler
```

However, I got the following error:

```
Traceback (most recent call last):
  File "./print_hierarchy.py", line 11, in <module>
    from systemrdl import RDLCompiler, RDLListener, RDLWalker, RDLCompileError
  File "/root/my/systemrdl-compiler/examples/../systemrdl/__init__.py", line 2, in <module>
    from .compiler import RDLCompiler
  File "/root/my/systemrdl-compiler/examples/../systemrdl/compiler.py", line 3, in <module>
    from antlr4 import CommonTokenStream
  File "/usr/local/lib/python3.4/dist-packages/antlr4/__init__.py", line 5, in <module>
    from antlr4.CommonTokenStream import CommonTokenStream
  File "/usr/local/lib/python3.4/dist-packages/antlr4/CommonTokenStream.py", line 33, in <module>
    from antlr4.Lexer import Lexer
  File "/usr/local/lib/python3.4/dist-packages/antlr4/Lexer.py", line 12, in <module>
    from typing.io import TextIO
ImportError: No module named 'typing'
```

The fix is trivial, but all the dependencies should be installed automatically:

```
python3 -m pip install systemrdl-compiler
```

amykyta3, 2018-11-15:
I dont officially support Python 3.4.

Part of the reason is because the Antlr parser I use depends on the "typing" library that was introduced in Python 3.5 (https://docs.python.org/3.5/whatsnew/3.5.html)

Looks like someone had made a backport of the typing module for older versions: https://pypi.org/project/typing/

Muffo, 2018-11-15:
I didn't realize typing was a built-in library in 3.5.
Thanks for the quick feedback.

amykyta3, 2018-11-16:
Fixed. See details in #16

closed 2018-11-16 by amykyta3