SystemRDL/systemrdl-compiler #15: ImportError: cannot import name 'messages'

Muffo, 2018-11-15:
After installing the library, I am trying to run the sample code.
From the repository root folder:

```
$ examples/print_hierarchy.py
Traceback (most recent call last):
  File "examples/print_hierarchy.py", line 11, in <module>
    from systemrdl import RDLCompiler, RDLListener, RDLWalker, RDLCompileError
  File "/root/my/systemrdl-compiler/examples/../systemrdl/__init__.py", line 2, in <module>
    from .compiler import RDLCompiler
  File "/root/my/systemrdl-compiler/examples/../systemrdl/compiler.py", line 5, in <module>
    from . import messages
  File "/root/my/systemrdl-compiler/examples/../systemrdl/messages.py", line 12, in <module>
    from .preprocessor.preprocessor import PreprocessedInputStream
  File "/root/my/systemrdl-compiler/examples/../systemrdl/preprocessor/preprocessor.py", line 11, in <module>
    from .. import messages
ImportError: cannot import name 'messages'
```

Is this a configuration issue or a bug in the sample code?

Commit id: a1303675eee8d309db78c2e4d18ff96da1051d60

amykyta3, 2018-11-15:
Local tests as well as seem to work.
I suspect this is also related to running using Python 3.4

If possible, see if you can use Python 3.5 or newer.
In the meantime, I'll setup up a virtualenv and see if I can work through getting 3.4 supported as it is still pretty common.

Muffo, 2018-11-15:
That's a good point!  I'll see if I can test this with a more recent version of Python.

amykyta3, 2018-11-16:
Fixed. See details in #16

closed 2018-11-16 by amykyta3