SystemRDL/systemrdl-compiler #25: The following code does not compile

hmohajeri, 2019-07-08:
I have this code from systemRDL document but cna't get it to compile and generates the following error.

error: Address map 'top' must contain at least one reg, regfile, mem, or addrmap.
fatal: Elaborate aborted due to previous errors

addrmap top {
     regfile example {
     reg some_reg { field {} a; };
     some_reg a  @0x0;
     some_reg b  @0x4;
     some_reg c;
     some_reg d @0x10;
 
   
};
};

PRWald, 2019-07-08:
I think you need to name the regfile. Try adding a name after the closing
brace of the regfile. The way I think of it, "example" is the class name
and the name after the closing brace is the instance name. Not sure if this
is the right way of thinking about it. Maybe someone on this list who knows
more could explain.

amykyta3, 2019-07-08:
@PRWald is correct. The example shown in the spec is a little misleading since they do not properly instantiate the regfile declared as `example`. Declaring the regfile definition only makes the type available in the local namespace. Since the example does not create an instance of the component, it does not exist, and ends up violating one of the semantic checks (components cannot be empty)

A corrected example would create an instance of `example` as shown here:

```
addrmap top {
    regfile example {
        reg some_reg { field {} a; };
        some_reg a @0x0;
        some_reg b @0x4;
        some_reg c;
        some_reg d @0x10;
    };
    example example_inst @ 0x0;
};
```

I already have a running list of typos/oversights the RDL spec authors missed here: https://github.com/SystemRDL/systemrdl-compiler/blob/master/docs/implementation_notes/systemrdl_spec_issues.md
I'll add this one to the list.

hmohajeri, 2019-07-09:
Many thanks for the help and appreciate it. The code now works beautifully. Amykyta3, thanks for the errata link.

closed 2019-07-09 by hmohajeri