SystemRDL/systemrdl-compiler #27: An error about "donttest"

you2quan, 2019-08-06:
The version of systemrdl-compiler is 1.71, a Error:
`
error: Property 'donttest' expects a boolean for non-field types. Got an integer in 'HPS_PAGE'
    } HPS_PAGE @ 0x008;
`
my rdl code is:
`    reg HPS_PAGE {
        desc = "Provides visibility of the page for host accesses to HPS. Previously this was a configurable value but is now fixed at FPGA image compile time";
        donttest = true; // exclude from plain simple hw register test
        stat_f hpsPage[31:21]; hpsPage->desc = "If applicable, the address presented to the ARM interface will be prepended by this value.";
    } HPS_PAGE @ 0x008;
`

amykyta3, 2019-08-07:
Thanks for reporting. I have confirmed the issue and will fix.

In the meantime, using an implicit true assignment should work:
```
reg HPS_PAGE {
  desc = "Provides visibility of the page for host accesses to HPS. Previously this was a configurable value but is now fixed at FPGA image compile time";
  donttest; // exclude from plain simple hw register test
  stat_f hpsPage[31:21];
  hpsPage->desc = "If applicable, the address presented to the ARM interface will be prepended by this value.";
} HPS_PAGE @ 0x008;
```

you2quan, 2019-08-07:
> Thanks for reporting. I have confirmed the issue and will fix.
>
> In the meantime, using an implicit true assignment should work:
>
> ```
> reg HPS_PAGE {
>   desc = "Provides visibility of the page for host [...]

thank you

amykyta3, 2019-08-07:
Fixed in v1.7.2

closed 2019-08-07 by amykyta3