SystemRDL/systemrdl-compiler #28: Enumerated types in user-defined properties

pdonahue-ventana, 2019-08-08:
Here is example 2 from section 15.2.2 of the SystemRDL 2.0 spec:
```
enum myEncoding {
  alpha = 1'b0;
  beta = 1'b1;
};
property my_enc_prop {
  type = myEncoding;
  component = field;
  default = beta;
};
addrmap top {
  reg {
    field { my_enc_prop = alpha ; } f ;
  } regA ;
} ;
```

When I try compiling that, I get an error:
```
uarch/lib/registers/foo.rdl:8:12: fatal: Reference to 'beta' not found
  default = beta;
            ^^^^
```

I'm extremely interested in having this functionality of user-defined properties which are enumerated types.

pdonahue-ventana, 2019-08-08:
I now saw line 72 of https://systemrdl-compiler.readthedocs.io/en/release-1.7/_downloads/31bdc46296fae3e927a5dd42b4709c3a/semantic_checks.ods

Perhaps this is a known limitation.  If so, consider this a vote to prioritize this when deciding which limitations to fix first.  Thanks!

amykyta3, 2019-08-09:
No bug here! You're just missing the type name from your enumeration references. From the SystemRDL spec, section 6.2.5.1:

> Enumerator
references shall be prefixed with their enumerated type name and two colons (`::`), e.g.,
`MyEnumeration::MyValue`

Changing your example to the following works correctly:
```
enum myEncoding {
  alpha = 1'b0;
  beta = 1'b1;
};
property my_enc_prop {
  type = myEncoding;
  component = field;
  default = myEncoding::beta;
};
addrmap top {
  reg {
    field { my_enc_prop = myEncoding::alpha ; } f ;
  } regA ;
};
```

closed 2019-08-09 by amykyta3

amykyta3, 2019-08-09:
I just realized you pulled this from the spec. I've found that not all their examples actually compile correctly and often violate their own rules. I have a running list of issues I've run into in their spec, including this one:
https://systemrdl-compiler.readthedocs.io/en/latest/dev_notes/rdl_spec_errata.html#example-2

pdonahue-ventana, 2019-08-09:
Spec bugs are always annoying, particularly to a relative newbie like me.  This problem wasn't as obvious to me as the missing semicolon in example 1 (which you have also noted on the page you cited).  When I noticed the missing semicolon I was wondering if there was a way to inform someone to get it fixed.  I'll assume that either the answer is essentially no or that you have done it.

Thanks for your help.  This tool is really great.

amykyta3, 2019-08-09:
Yep! That's the motivation for that page. I've raised some of the issues with the committee already, and they should be aware of the page I linked. Generally the motivation to update was low since they had already released the spec, and I'm just some random dude on the internet (shrug).

Glad to hear people are using the compiler!