SystemRDL/systemrdl-compiler #30: User defined property default values not propogated

bdmagnuson, 2019-08-14:
When executing node.get_property(<user prop>) the call returns None unless the property was explicitly given a value despite the property being defined with a default.

amykyta3, 2019-09-02:
Sorry I didnt respond sooner. I've been exceptionally busy the past month.

According to the [SystemRDL 2.0 spec](https://www.accellera.org/downloads/standards/systemrdl) this is actually the intended behavior. (I'm not a huge fan of it either, but that's what the spec says...)

There are several places in the spec that describe this behavior.

Intro paragraph in section 15:
> Unlike built-in properties, user-defined properties are not automatically bound to every instance of the specified component’s type; instead they need to be bound per instance and/or component definition.

**So what's the point of specifying `default`?**

According to the spec, the `default` value used when binding the property to component without a value.

Here's a trimmed down version of the example pulled from 15.2.2-Example1:

```
property some_bool_p { type = boolean; component = field; default = false; };

addrmap foo {
    reg{
        field { some_bool_p; } field1;  // Attach some_bool_p to the field
                                        // with a value of false;

        field { some_bool_p = true; } field2;
        // Attach some_bool_p to the field with a value of true;

    } bar;
};
```
For normal built-in properties, a no-value assignment implies the property was assigned `true`. For UDPs, it implies the assignment of its default. Notice how `field1` in the example inherits the default value of false for the `some_bool_p` property. `field2` is explicitly assigned the value `true`.

**So why `None`?**

This goes back to the original text in the spec I referenced. If a UDP was never bound to a component, it will not have a value. Therefore calling `get_property()` returns `None`.

bdmagnuson, 2019-09-03:
Bummer.  That is a shame.  Thanks for the detailed response to my very brief report.

closed 2019-09-03 by bdmagnuson