SystemRDL/systemrdl-compiler #34: RDLCompiler parse errors

sjalloq, 2019-10-15:
Hi

I wanted to try out your compiler but it seems to fall over with the syntax I'm using.  So far I've used the Juniper ORDT tools without any issues and they process this RDL fine.  It could be the naming convention I used to create easier to read register names using a double underscore.  I tend to use "reg_name__field_name" style.

` regfile.rdl.pp.final:580:23: error: no viable alternative at input '_sff_irq_flags._irq_3.*'
	_sff_irq_flags._irq_3.*->enable = _sff_irq_en_flags._irq_en_3.*;` 

The full RDL is attached.

[rdl.txt](https://github.com/SystemRDL/systemrdl-compiler/files/3729990/rdl.txt)

amykyta3, 2019-10-16:
There's nothing wrong with your naming convention - underscores can be used any way you want in identifiers.

It looks like you're using the wildcard assignment syntax that is unique to ORDT:
```
_sff_irq_flags._irq_3.*->enable = _sff_irq_en_flags._irq_en_3.* ;
                      ^                                       ^
```
See [ORDT's Wiki for more details](https://github.com/sdnellen/open-register-design-tool/wiki/Using-SystemRDL-input#ordt-syntax-differences-vs-systemrdl). There are several non-standard extensions to the language that were added in that tool.

Assigning instance properties using a wildcard is not part of Accellera's SystemRDL standard, so I did not implement it.

sjalloq, 2019-10-16:
Ah, of course.  Sorry.  What's the best way of fixing this?  Use the Perl preprocessor to expand a loop?

amykyta3, 2019-10-17:
If sticking strictly to the RDL spec, then you'd have to explicitly assign each. Your IRQ 14 for example:

```
_sff_irq_flags._irq_14._tx4_pwr_low_warn->enable = _sff_irq_en_flags._irq_en_14._tx4_pwr_low_warn ;
_sff_irq_flags._irq_14._tx4_pwr_high_warn->enable = _sff_irq_en_flags._irq_en_14._tx4_pwr_high_warn ;
_sff_irq_flags._irq_14._tx4_pwr_low_alarm->enable = _sff_irq_en_flags._irq_en_14._tx4_pwr_low_alarm ;
_sff_irq_flags._irq_14._tx4_pwr_high_alarm->enable = _sff_irq_en_flags._irq_en_14._tx4_pwr_high_alarm ;
_sff_irq_flags._irq_14._tx3_pwr_low_warn->enable = _sff_irq_en_flags._irq_en_14._tx3_pwr_low_warn ;
_sff_irq_flags._irq_14._tx3_pwr_high_warn->enable = _sff_irq_en_flags._irq_en_14._tx3_pwr_high_warn ;
_sff_irq_flags._irq_14._tx3_pwr_low_alarm->enable = _sff_irq_en_flags._irq_en_14._tx3_pwr_low_alarm ;
_sff_irq_flags._irq_14._tx3_pwr_high_alarm->enable = _sff_irq_en_flags._irq_en_14._tx3_pwr_high_alarm ;
```

The perl preprocessor might help since a lot of the field names are similar, so at least you could remove some of the repetition. I can see why this was added as an extension in ORDT's implementation.

closed 2019-10-22 by amykyta3