SystemRDL/systemrdl-compiler #35: RuntimeError when assigning enum to reset property

bdmagnuson, 2019-10-16:
This code takes a RuntimeError trying to validate the RHS as an int or field reference.

```
enum test_e {
   VAL0 = 0;
   VAL1 = 1;
};

addrmap test_a {
   reg {
      field {
         encode = test_e;
         reset = test_e::VAL1;
      } test_f;
   } test_r;
};

```
Interestingly making the RHS an expression `test_e::VAL1 + 0` forces it to be evaluated down to an int before the validation check is done and works.  Which is an odd looking but effective workaround.

amykyta3, 2019-10-17:
Wow looks like I completely overlooked implicit type-casting for property assignments. I handle it correctly everywhere else (parameters, structs, enums...).
Thank you for finding and reporting this.

Your workaround works because the addition operator forces the enum to be cast to an integer during evaluation. The following will work too:
```
reset = longint'(test_e::VAL1);
```

Regardless, I'll work on a fix for this.

amykyta3, 2019-10-17:
Another workaround that is less ugly is to use the equivalent reset value assignment syntax:
```
field {
    encode = test_e;
} test_f = test_e::VAL1;
```

amykyta3, 2019-10-17:
Fixed. Will be released in upcoming v1.9.0.

In the meantime, you can install the dev version using:
```
python3 -m pip install --user -U git+https://github.com/SystemRDL/systemrdl-compiler.git
```

closed 2019-10-17 by amykyta3