SystemRDL/systemrdl-compiler #37: Fatal when importing Perl packages

RasmusGOlsen, 2019-12-04:
I'm getting the following error when using Perl packages that use the Perl module Exporters.

```
test.rdl: fatal: Encountered a Perl syntax error while executing embedded Perl preprocessor commands:
'sort' trapped by operation mask at /usr/share/perl5/vendor_perl/Exporter/Heavy.pm line 190, <> line 4.
Compilation failed in require at /usr/share/perl5/vendor_perl/Exporter.pm line 16, <> line 4.
BEGIN failed--compilation aborted at (eval 5) line 1, <> line 4.
```
Could the Perl safe compartment in the Perl preprocessor be expanded to permit `sort`?

I would like to import the variable names into the namespace instead of referencing the package like this `$parameters::top`

below is a short example as an illustration.

**parameters.pm**
```perl
package parameters;

use Exporter qw(import);

our @EXPORT = qw($top);
our $top = "MyTop";
```
**test.rdl**
```systemrdl
<% use parameters; %>

addrmap <%=$top%> {
    reg {
        field {} myfield;
    } myreg;
};

```

amykyta3, 2019-12-14:
Thanks for reporting.
Its difficult fore me to draw the line for which Perl opcodes to allow. On one hand, I want the Perl preprocessor to be useful, but on the other hand it makes sense to restrict access to some of the language for security/abuse reasons.

I'll re-visit the list of default allowed opcodes and see if they should be revised. Also, I may expose the ability to change them as an argument to the compiler. Currently they are hardcoded: https://github.com/SystemRDL/systemrdl-compiler/blob/master/systemrdl/preprocessor/ppp_runner.pl#L37

Here is the list of Perl opcodes I'm referring to: https://perldoc.perl.org/Opcode.html#Predefined-Opcode-Tags

I admit I am far from a Perl expert, so if you have recommendations on what opcode groups to allow, it would be very helpful.

RasmusGOlsen, 2019-12-14:
I think the best solution is to expose the ability to change them as arguments to the compiler. You will never find one set of opcodes that will suit everybody's needs.

amykyta3, 2019-12-19:
I re-reviewed the list of opcodes I was loading and I definitely missed a few, specifically some of the ones in the `:still_to_be_decided` group such as `safe`. Many of them should have been enabled before so I have added them to the default opcode list. In the next release, your preprocessor commands should work without needing to override the opcodes yourself.

Regardless, I have exposed an override flag in case it is needed in the future since I agree, it is good to have the option to change it.

Before I publish the release, I had one related question:

What are your thoughts on how module search paths are exposed to Perl? Currently I do not add anything to the module search path, so the compiler relies on the system paths, and whatever may be set in the `PERLLIB` environment variable.
I think it would be useful to expose the ability to provide additional Perl include paths from within the compiler. Do you think it would make sense to use the same include paths as specified by `incl_search_paths` as specified by [Compiler.compile_file()](https://systemrdl-compiler.readthedocs.io/en/release-1.9/api/compiler.html#systemrdl.RDLCompiler.compile_file)? Or should it be a separate input? (Currently `incl_search_paths` is only used for verilog-style includes)

RasmusGOlsen, 2019-12-19:
For Perl module search paths I think you should rely on how the Perl environment is set up. You can always add a module search path from within your Python script by setting the `PERLLIB` environment variable.
```python
import os
os.environ['PERLLIB'] = './mymodules'
```
If you decide to expose the ability to add module search paths from within the compiler I think it should be separated from the Verilog include paths.

amykyta3, 2019-12-20:
Good point. No reason to add a second mechanism to manipulate `PERLLIB`.

Released v1.10.0 just now. Should be fixed.

closed 2019-12-20 by amykyta3