SystemRDL/systemrdl-compiler #38: Address map nesting using the compiler

markcohen32, 2019-12-17:
Hi,
After reading all the relevant documentation, I still haven't found an explanation how to nest one address map within another one by using the compiler. In order to build more complex register documentation, this feature is needed. In other words, my question is how to 'import' one address map from one SystemRDL file to another one, and compile the new file?

RasmusGOlsen, 2019-12-17:
You will have to give a list of files to the compiler.

```python
import sys
import os
from systemrdl import RDLCompiler, RDLWalker, RDLListener, RDLCompileError

input_files = sys.argv[1:]
incdir = os.getcwd()

rdlc = RDLCompiler()
try:
    for input_file in input_files:
        rdlc.compile_file(input_file, incl_search_paths=[incdir])
    root = rdlc.elaborate()
except RDLCompileError as e:
    raise e
```

markcohen32, 2019-12-17:
I added the code, but still- while trying to print out the hierarchy, only the last file's output is presented.

RasmusGOlsen, 2019-12-17:
Can you show an example of the SystemRDL code and the code you use to walk the address map?

markcohen32, 2019-12-19:
Using the 'print_hirearchy.py' example that is given.
The SystemRDL code:

`      addrmap DMA_EX {

            name="DMA BLOCK REGISTER MAP";
            desc="this register list contains all the dma's registers";

            default regwidth=32;
            default hw=rw;
            default sw=r;

            regfile dma_ex{

            alignment=8;

            reg{
                field{
                name="reset value";

                }rst_value=0;

            } mps @0x0000;

            reg{
                field{
                name="src_dfm_max_rd_burst";
                desc="source dfm max read burst length";
                }name;
            } src_dfm_max_rd_burst @0x8;

               reg {
                field{
                name="src_dfm_min_rd_burst";
                desc="source dfm min read burst length";
                }name;
            } src_dfm_min_rd_burst;

                reg{
                field{
                name="dbg_src_eoo_stream_cnt";
                desc="source eoo streamed out count";
                }name;
                }dbg_src_eoo_stream_cnt[2];
             }dma_reg_file;

             regfile debug_events_dma{

                 reg source_description{
                    regwidth=256;
                     field{
                         name="reset value";

                     }rst_value=0;

                  } src_dsc;

                reg{
                regwidth=64;
                 field{
                     name="reset value";

                      }rst_value;

                } src_ctrl;

             }dbg_events_dma;

             addrmap IDENTIFY{

            name="IDENTIFY BLOCK REGISTER MAP";
            desc="this register list contains all the identify's registers";
            bridge;
            default regwidth=32;
            default hw=rw;
            default sw=r;

           reg dubug_current_id_cmd1{
                field{
                    name="reset value";
                }rst_value=0;
            }dbg_curr_id_cmd1 @0x0;

         };

            };
`
The compiler ignores the second address map "IDENTIFY" which is wrapped in another address map.
My main goal is to define and characterize different address maps in different .rdl files and nest them one in another to create a larger,more complex maps by importing them to other .rdl files. Thus, I won't need to re-write them again and again. Would like to have some assistance with that, if possible.

amykyta3, 2019-12-19:
I see whats happening. The thing you are missing is the instantiation of the IDENTIFY addrmap.
You have declared the `IDENTIFY` component type, but it was never instantiated:

```
IDENTIFY identify_instance @ 0x1000; // (or whatever address you wanted it at)
```

Assuming you wanted to split between files, you could do the following:
`identify.rdl`:
```
addrmap IDENTIFY {
    // (contents not shown)
};
```

`dma_ex.rdl`
```
addrmap DMA_EX {
    // (other contents not shown)
    IDENTIFY my_identify_block @ 0x1000;
};
```

RasmusGOlsen, 2019-12-19:
Think about it as Verilog modules or VHDL entities. You declare the component e.g. `addrmap` and to use it, it needs to be instantiated. It is only the top component that doesn't need to be instantiated. In SystemRDL you can also do the declaration and instantiation at the same time. Then it will look like this. 

```systemrdl
addmap DMA_EX {
    // (contents not shown)
    addrmap IDENTIFY {
        // (other contents not shown)
    } my_identify_block @ 0x1000;
};
``` 
You would probably place each `addrmap` in separate files as you describe, but you are already using the direct declaration and instatiation for your register files, registers and fields.

amykyta3, 2020-01-01:
Closing since it looks like the question was answered. @markcohen32 feel free to re-open this if you still need help.

closed 2020-01-01 by amykyta3