SystemRDL/systemrdl-compiler #40: Singlepulse and onwrite field's software access properties activated at the same time

laumonte10, 2020-01-13:
Hello,

I have read in the Systemrdl documentation about singlepulse and onwrite field's software access properties and for me it seems that if you activate the singlepulse property in a field you couldn't have an onwrite property in the same field. Because both of them change the content of the field.

I was playing around with the systemrdl-compiler and it seems that It is allowed to have both properties specified in the same field. Example:

field Fld4{onwrite = wclr;
                singlepulse = true;
                     } 

I am wondering if this is the correct behaviour or the compiler shouldn't allowed this.

Thank you very much.

Laura.

amykyta3, 2020-01-14:
Thank you for reporting!
Although the SystemRDL spec doesn't explicitly say this, I agree that using `singlepulse` and `onwrite` properties together shouldn't be allowed. You are correct that if a field has `singlepulse==true`, then assigning any value to the `onwrite` property does not make sense since their behaviors would conflict.

If you can think of any other illegal/conflicting property combinations, I'd be happy to add them to the validation checks. Since there are countless combinations possible, the official spec only describes the more common conflicts.

laumonte10, 2020-01-14:
There are two other combinations that shouldn't be allowed: _singlepulse_ and _woset_ and the other one is _singlepulse_ and _woclr_. These are the two write properties that are also included in the _onwrite_ property.

For the moment I can not think in any other illegal property combination. 
Thank you.

amykyta3, 2020-01-17:
Fixed and will publish in upcoming release.

I decided to only error if `singlepulse` is used alongside `wclr` or `woclr` since these are the only two that actually have conflicting behavior.
Even though it would be strange to use any of the other `onwrite` properties, they do not necessarily semantically conflict with `singlepulse`.

Thanks again for submitting!

closed 2020-01-17 by amykyta3