SystemRDL/systemrdl-compiler #42: Ability to have parametric number of component

markcohen32, 2020-03-05:
Hi,

 Is there an option to assign a parametric value to an instance of components such as regfile 
, addressmap or reg, without initializing a value of `enum`?
Further more, is it possible to assign a value to a component from another component's property or even defining a variable that will take some value ? for instance:

`
enum counting{
               regnum=a;   //some variable with the value 'a' that will be defined later or will take other 
                                      component's property value (such as size).
                         };

addrmap TOP{

    default regwidth=32;
    alignment=64;
    reg reg_constructor #(longint unsigned SIZE=5){
        field{
            name="original name";
            desc="some description for this field";
            hdl_path_slice='{"rtl_f1_1","rtl_f1_0"};
        } f1[1:0];
        field{
            hdl_path_slice='{"rtl_f2"};
        } f2[5:3];
    };

    reg_constructor reg1[counting::regnum]; //regnum=a;
`

For now, such code doesn't compile.
Would like to have some advice in this issue.
Thanks!

amykyta3, 2020-03-06:
Sounds like there are two separate questions here.
Using parametric values is supported via component parameters. For example:
```
addrmap top #(longint unsigned N=4) {
    reg my_reg_t {
        field {} f;
    };

    my_reg_t reg_array[N];
};
```
In the above example, `reg_array` defaults to have 4 elements, but if needed, could be overridden to any other value via the parameter. 

Unfortunately, as the spec is currently written, it doesn't allow for parameters to be used in a definition such as an enum (due to clause 5.1.1.2-d). For example, this would not be allowed due to the current namespace rules:
```
addrmap top #(longint unsigned N=4) {
    enum my_enum {
        abcd = N;
    };
};
```
However I'm starting to question whether the spec authors even know about their own rule described in 5.1.1.2-d since they seem to violate it all the time [in their examples](https://systemrdl-compiler.readthedocs.io/en/latest/dev_notes/rdl_spec_errata.html#example-in-6-2-6-uses-an-invalid-hierarchical-reference).

Did that answer your question? If not, maybe a more specific use-case would help me understand what you are trying to do.

markcohen32, 2020-03-08:
First of all, thanks for the response!
I'm trying to assign a value to a parameter dynamically. For design and verification purposes, it's necessary for me to be able to override some values while using some of the design in higher\ lower levels.
Let's look at the example you used in your comment. Is it possible to change the value of 'N' in another part of the code? like:

```

addrmap top #(longint unsigned N=4) {
    reg my_reg_t {
        field {} f;
    };

    my_reg_t reg_array[N];
    
    N=7;
    my_reg_t new_reg_array[N];

   
};
```

Or, assigning 'N' with a value that comes from another component's property? using the same example:

```
addrmap top #(longint unsigned N=4) {
    reg my_reg_t {
        field {} f;
    };
    
    N=my_reg_t.f->reset;
    my_reg_t reg_array[N];
};
```

In this example, I'm trying to assign 'N' with the reset value of the field 'f'.

amykyta3, 2020-03-10:
I realize I did not complete my example from before.
If you want to re-use a component with different parameterizations in RDL, you can do so by overriding the parameter during instantiation. Consider the addrmap `top` from before - I can instantiate multiple variants of it, each using different values for N:

```
addrmap system_top {
    top top1; // Uses default N=4
    top #(.N(8)) top2; // Overrides N=8
};
```
Also, remember that parameters are constants and are resolved during compile-time. It would not make much sense for something like `reg_array[N]` to change size at runtime since it describes hardware in FPGA or ASIC.

Lastly regarding this snippet:
```
N=my_reg_t.f->reset;
```
There are some obscure cases where you can reference an RDL property, but it is only to describe a structural connection in the hardware. Probably not what you were looking for :-)

closed 2020-03-23 by amykyta3