SystemRDL/systemrdl-compiler #45: Query the available user defined properties

tolaszi, 2020-03-17:
I was looking at the user defined properties and by the UDP tests and examples it is quite clear how to check a UDP value on an instance. However, I was wondering whether there is built-in mechanism to query all the UDPs (and their attributes) that are defined within a SystemRDL file?

I also would like to point out one minor inconsistency I stumbled upon. The following code:

```
property bool_prop {type = boolean; component=field; default=false;};
addrmap test {
    reg {
        field {
            bool_prop;
        } FIELD[4] = 0;
    } TEST @ 0x0;
};
```

would return `0` when doing a `print(node.get_property("bool_prop"))` on the Python side (and `1` when the default is set to `true`. However, when the UDP value is set in the field definition:
```
        field {
            bool_prop=true;
        } FIELD[4] = 0;
```

the same query returns `True/False`. It is no big deal, equality checks work just the same, just thought I'd let you know.

amykyta3, 2020-03-19:
I dont have anything built-in that would query user-properties directly like that. 
This isn't documented, but you could fetch the names of all UDPs that exist in the current design:
```python
rdlc = RDLCompiler()
# [compile & elaborate as usual...]

all_udp_names = list(rdlc.env.property_rules.user_properties.keys())
```
Perhaps I'll break that out into a more official method since that seems useful.

Combining the above with the [Node.descendants()](https://systemrdl-compiler.readthedocs.io/en/latest/api/node.html#systemrdl.node.Node.descendants) method to traverse all child instances, you could probably accomplish what you want.

Regarding the boolean getting cast to a 1/0: I'll look into that. Officially, the RDL spec says that `true` and `false` are simply aliases for `1` and `0`, but I've been striving to preserve the boolean type whenever possible since it's less surprising to the user.

tolaszi, 2020-03-19:
Thanks a lot for the pointer! With this I could query the information I was looking for.

amykyta3, 2020-03-23:
Added some more mechanisms to list UDPs in the upcoming v1.12 release:
* Added [RDLCompiler.list_udps()](https://systemrdl-compiler.readthedocs.io/en/latest/api/compiler.html#systemrdl.RDLCompiler.list_udps)
* Added more options to [Node.list_properties()](https://systemrdl-compiler.readthedocs.io/en/latest/api/node.html#systemrdl.node.Node.list_properties)

Also fixed the boolean issue.
Thank you again for reporting!

closed 2020-03-23 by amykyta3