SystemRDL/systemrdl-compiler #46: The ANTLR license

martinda, 2020-03-22:
Why do you copy the ANTLR license?
I would also like to draw your attention to the fact that the ANTLR license is a [modified 3-clause BSD license](https://groups.google.com/d/msg/antlr-discussion/lxBXsqXcRks/pIi4w6A0CQAJ) which protects it author but not its contributors. So is it really required here? If you really need the BSD license I suggest you use the real one, not the ANTLR one.

amykyta3, 2020-03-22:
This project re-distributes the Antlr C++ runtime source (Python sdist packages), as well as in binary form (Python wheels) on [PyPi](https://pypi.org/project/systemrdl-compiler/#files).
[Clauses 1 and 2 in Antlr's license](https://github.com/antlr/antlr4/blob/master/LICENSE.txt#L8-L12) clearly state the terms that re-distributions shall also include proper attribution, along with the rest of the license.

Although you are correct that the text Antlr uses for its license does not match the latest recommended 3-clause BSD text, it would be inappropriate for me to modify the Antlr license without the author's explicit permission.

Keep in mind that the 3rd-party attribution license for the `systemrdl-compiler` project only applies to the Antlr files in [this directory](https://github.com/SystemRDL/systemrdl-compiler/tree/master/systemrdl/parser/ext/antlr4-cpp-runtime). The rest of this project is licensed under a very permissive [MIT license](https://github.com/SystemRDL/systemrdl-compiler/blob/master/LICENSE) and any contributions to this project would be covered by that.

martinda, 2020-03-23:
> Keep in mind that the 3rd-party attribution license for the `systemrdl-compiler` project only applies to the Antlr files in this directory.

Thank you. That was not clear to me. Do you think you should be adding an explanation as to what each license applies to in the README?

I have many more questions.

Why the is antlr source redistributed? Probably for convenience I guess.

But more importantly, what happens if someone modifies the `systemrdl-compiler` local copy of the antlr sources? Do they become contributors to ANTLR? If so, there is the [contributors.txt](https://github.com/antlr/antlr4/blob/master/contributors.txt#L32-L37) agreement, paragraph 3, which deserves a very close read: contributors agree to indemnify the "ANTLR Project and its users" from harm, quite broad IMO. I've never seen such broad clause in other open source projects.

How would you handle PRs that aim to change the ANTLR sources?

Thanks in advance for clarifying all of this.

amykyta3, 2020-03-23:
The Antlr license is intentionally kept separate in a `LICENSE-3RD-PARTY` file. General GitHub convention is to do so, with a `LICENSE` file denoting the project's primary license. Now that you mention it, I don't explicitly point to the license anywhere in my readme or documentation so I can add that clarification.

> Why the is antlr source redistributed?

Originally I primarily used the ANTLR Python runtime library. This would get installed as [a separate dependency](https://github.com/SystemRDL/systemrdl-compiler/blob/master/setup.py#L70) from [PyPi](https://pypi.org/project/antlr4-python3-runtime/), so it did not have any reason to be bundled into this project.
Recently, I wrote a C++ based accelerator using the ANTLR C++ runtime. In order to seamlessly distribute it, I have to include a copy of it within my project.

> what happens if someone modifies the `systemrdl-compiler` local copy of the antlr sources?

I don't interpret that as relevant in this case. For one to be a contributor of the Antlr project, you would have to apply a change to the original Antlr codebase. People fork repositories to patch in their niche changes all the time. I'd personally never consider them contributors unless their change was intentionally merged back into the mainline project. In fact, you could interpret them as authors of their own modifications in the interim.

> contributors agree to indemnify the "ANTLR Project and its users" from harm, quite broad IMO. I've never seen such broad clause in other open source projects.

You're probably overthinking it. I'm interpreting this as the "Don't be a jerk, and it's not my problem if you're a jerk" clause. The open source community encourages everyone to participate in free software. Of course that opens you up to all kinds of folks - some come with strong opinions, immense egos, unreasonable grudges, etc... The only right their `contributors.txt` file grants is the right to submit code to the Antlr project. If two potential contributors get into a disagreement, and go as far as manipulate the other's ability to participate in the community, it would be a violation of this right, and also not the responsibility of the Antlr folks.

> How would you handle PRs that aim to change the ANTLR sources?

I would reject them :stuck_out_tongue_closed_eyes: . I have no interest in modifying the Antlr runtime. Forking a project needs some serious justification since you're effectively on your own to maintain those changes. The only pull request I'd consider is if it is updating the runtime to a newer official version. Even then I'd probably reject it and do it myself - more for security reasons than licensing.

amykyta3, 2020-04-07:
Closing for now. If you have any additional questions, feel free to re-open this.

closed 2020-04-07 by amykyta3