SystemRDL/systemrdl-compiler #47: Implement SystemRDL Constraints

amykyta3, 2020-03-23:
SystemRDL 2.0's constraints is the last major feature that is currently unimplemented. Currently, any RDL that contains a constraint block will fail to compile.

* An elaborated constraint is exposed to the user as an abstract syntax tree (AST)
* Since most users will simply want to dump the constraint to SystemVerilog, provide a built-in framework to do so.
  * Will require a mechanism for user-defined reference resolution.
* Provide some level of constraint validation.

hgdangkhoi, 2021-05-25:
Do you have an ETA on when this will be implemented? Thanks!

amykyta3, 2021-05-26:
Not sure. This feature is a pretty low priority for me.

Samrat-03, 2025-01-23:
Is this feature implemented now? With the current version, if the rdl has constraints, the compiler doesn't show any error. But PeakRDL doesn't reflect it in uvm regmodel. Is this PeakRDL issue of SystemRDL Compiler issue?

jeras, 2025-02-19:
While SystemRDL-compiler is the parser, PeakRDL is a code generator using SystemRDL-compiler as a library (Python module). In this case I would say, the PeakRDL regmodel tool does not have the code to generate constraints. Also based on this thread, even if the compiler does not crash, it does not seem constraints are properly processed by the parser. In short the feature is not implemented yet.