SystemRDL/systemrdl-compiler #49: Traversing the model

markcohen32, 2020-04-06:
Hi,
I'm currently writing a JSON exporter for My designs. I have based my script on the Listener and Walker that are given with the RDL compiler. 
The issue is that whenever i'm trying to run the code on an address map, it seems that only the last register of the design is visited. Maybe I'm missing something.
Would be great to have some advise. Here's the code:

`

import json
import compilation
from systemrdl import RDLListener
from systemrdl import  RDLWalker
 from systemrdl.node import  AddrmapNode, RegNode, AddressableNode

class JSONListener(RDLListener):
        def __init__(self,mydict):
            super().__init__()
            self.main_dict =mydict
            main_dict['Address Map']=None

        def enter_Addrmap(self, node):
            map_size = node.size
            map_name = node.inst_name
            if isinstance(node,AddrmapNode):
                map_address_absolute=node.absolute_address
                map_address_offset=node.address_offset
                addrmap_dict = {'Addressmap Name': map_name,
                                'Map Size': map_size,
                                'Map offset address': map_address_offset,
                                'Map absolute address':map_address_absolute,
                                 }
                main_dict['Address Map']=addrmap_dict

        def enter_Field(self, node):
                bit_range_str = "[%d:%d]" % (node.high, node.low)
                sw_access_str =  node.get_property("sw").name
                hw_access_str =  node.get_property("hw").name
                design_params =  node.get_property("design_params")
                field_width = node.width
                field_dict= {'sw': sw_access_str,
                             'hw': hw_access_str,
                             'bit range': bit_range_str,
                             'field width': field_width,
                             'design params': design_params}
                main_dict['Address Map']['Registers']['Fields']=field_dict

        def enter_Reg(self, node):
                reg_dict={}

                if isinstance(node,RegNode):
                    reg_dict.update([('Reg', node.get_path_segment())])
                    reg_size = node.size
                    reg_address_absolute=node.absolute_address
                    reg_address_offset=node.address_offset
                    reg_dict.update([('Reg Size', reg_size),
                                     ('Address Offset', reg_address_offset),
                                     ('Address Absolute', reg_address_absolute),
                                     ('Instantiated From',node.orig_type_name),
                                      ('Fields',None)])
                    main_dict['Address Map']['Registers']=reg_dict

walker = RDLWalker(unroll=True)
main_dict={}
listener = JSONListener(main_dict)
walker.walk(compilation.root, listener)

print(json.dumps(main_dict))

`

My thought was to use the Listener in order to create a dictionary that will be nested to an upper-level dictionary. Thus, creating a modular data structure. 
Thanks in advance. You're doing a great job!

amykyta3, 2020-04-07:
It looks like in `enter_Reg()`, you're always creating a fresh `reg_dict` dictionary and entirely replacing whatever you had in the `main_dict` data structure every time that function is called.

In this case, I'd actually recommend against using the walker/listener traversal method. The walker/listener is really only useful if you want to traverse the register model, but don't really care much about keeping track of hierarchy. In other words, the walker/listener method is great if you want to convert hierarchical -> flat data. Its good for quick report generation, or if you just want to traverse everything to collect some data.

In your case, you want to generate structured JSON output, which is a hierarchical -> hierarchical translation. You're probably better off using several recursive functions since it will naturally result in the structure you want. Here is a quick example to get you started:

```python
class JSONConverter:

    def convert_addrmap(self, addrmap):
        # Initialize a new dictionary that represents the current `addrmap`
        addrmap_dict = {}
        addrmap_dict['name'] = addrmap.get_path_segment()

        # Collect a list of dictionaries for all children
        children = []
        for child in addrmap.children(unroll=True):
            if isinstance(child, AddrmapNode):
                child_dict = self.convert_addrmap(child)
                children.append(child_dict)
            elif isinstance(child, RegNode):
                child_dict = self.convert_reg(child)
                children.append(child_dict)
            else:
                print("TODO: still need to translate:", child)
        
        addrmap_dict['children'] = children

        # return the dictionary so that parent nodes (or the top-level) recieve
        # the completed data structure.
        return addrmap_dict

    def convert_reg(self, reg):
        reg_dict = {}
        reg_dict['name'] = reg.get_path_segment()

        reg_dict['fields'] = []
        for field in reg.fields():
            reg_dict['fields'].append(self.convert_field(field))
        
        return reg_dict

    def convert_field(self, field):
        field_dict = {}

        field_dict['name'] = field.inst_name
        field_dict['sw'] = field.get_property('sw').name
        field_dict['hw'] = field.get_property('hw').name

        return field_dict

jc = JSONConverter()
main_dict = jc.convert_addrmap(compilation.root)
print(json.dumps(main_dict))
```

The `convert_addrmap()` function is recursive since you could have multiple nested within each other in RDL. The addrmap and reg converter functions both traverse their immediate children, and collect their results into a list. Each `convert_*` function returns a dictionary that represents the object being visited.
Along the way, you can add the pertinent information into each node's dictionary and it will end up in the JSON output.
I didn't handle all node types (regfile, mem, signal...), but you should be able to get the idea on how to expand this.

markcohen32, 2020-04-08:
Thank you very much for the detailed explanation and the example!

closed 2020-04-11 by amykyta3