SystemRDL/systemrdl-compiler #50: Elaborating multiple root-level addremaps

markcohen32, 2020-04-16:
Hi.

So far, I've been using the following script in order to elaborate the design, assuming that the function gets a file with a list of .rdl files:

`def compiling_file(input_files):
    # Collect input files from the command line arguments: input_files=sys.argv[1:].
    # Now, the wrapping script collects the files from the command line and pass it as a list to the compiling_file func.

    #print(input_files)
    incdir = os.getcwd()  # addition for compiling more than one file
    # Create an instance of the compiler
    rdlc = RDLCompiler()
    rdlc.define_udp("design_params", str, [Addrmap, Regfile, Field], None)
    roots=[]
    try:
        # Compile all the files provided
        for input_file in input_files:
            rdlc.compile_file(input_file, incl_search_paths=[incdir])

        # Elaborate the design
        roots.append(rdlc.elaborate())
        return roots
    except RDLCompileError:
        # A compilation error occurred. Exit with error code
        sys.exit(1)`

As you can see, if the function gets multiple files that each one of them contains a list of .rdl files representing addrmaps nested one within another, we'll get a list of RootNodes to work with.
My question- what if I want to pass a list of files containing more than one Root in order to get the roots of all of them and not only the last one?
for example, assume that we have a file containing the following .rdl files:
/current_dir/addrmap1.rdl
/current_dir/addrmap_nested_in_1.rdl
/current_dir/addrmap2.rdl
 But, there's no interaction between the firsts and the last one. How can I use the .elaborate() function to get 2 RootNodes?

amykyta3, 2020-04-17:
The `rdlc.elaborate()` function will always build one register model tree at a time.
Given no parameters, it will make its best guess at which top-level definition you want and choose the last one defined. (See documentation: https://systemrdl-compiler.readthedocs.io/en/latest/api/compiler.html#systemrdl.RDLCompiler.elaborate)

In your case, you'll want to be explicit, and elaborate multiple register model trees out of the same compilation namespace. You can tell `elaborate()` exactly which addrmap type names you want:

```python
root1 = rdlc.elaborate("my_addrmap1")
root2 = rdlc.elaborate("my_addrmap2")
```

Your function wont know what top-levels it wants on its own, so you may need to specify another parameter to it.

markcohen32, 2020-04-30:
Okay. If, for example, I would like to return more than one RootNode from a .rdl files who has 2 or more addrmaps within it that are not nested?
`addrmap map_1{
//some components
};

addrmap map_2{
//some components
};
`
Is there a way to do so using the .elaborate() function?
I'm willing to build some complex designs and not necessarily implement single hierarchy in one .rdl file, as shown above in the example . I'd like to use the compiler in such way that I won't need to separate each map a file of its own.

amykyta3, 2020-05-02:
As shown in my previous response, you simply call `elaborate()` twice:

```python
root1 = rdlc.elaborate("map_1")
root2 = rdlc.elaborate("map_2")
```

markcohen32, 2020-05-02:
I meant, without the knowledge of what maps I have in this file, is there a way to return more than one root and not just the last one?

amykyta3, 2020-05-02:
Ah in understand now.
You could first query for all addrmap definitions in the root namespace, then call `elaborate()` multiple times to collect the elaborated trees of each.

```python
from systemrdl.component import Addrmap

# ... compile all input files as usual ...

# Collect names of all addrmap components defined in the root namespace
root_addrmap_names = []
for type_name, obj in rdlc.root.comp_defs.items():
    # Skip anything that isn't an addrmap, since we can't elaborate other stuff as root nodes anyways.
    print(type_name)
    if isinstance(obj, Addrmap):
        root_addrmap_names.append(type_name)

# Elaborate all of them
elaborated_addrmaps = []
for type_name in root_addrmap_names:
    root = rdlc.elaborate(type_name)
    elaborated_addrmaps.append(root)
```

Note that this will elaborate *all* addrmaps in the root namespace. If that's what you want, then great. But you may find yourself filtering out objects as your design grows and starts nesting addrmaps.

closed 2020-05-29 by amykyta3