SystemRDL/systemrdl-compiler #52: Reg inside mem

hgdangkhoi, 2020-05-06:
Hello,

First of all would like to say thank you for working on this project. It is really helpful. I have a quick question, consider the addrmap below:
```
 addrmap test#(longint unsigned BANKS=2) {
  name  = "test Address Map";
  hdl_path = "test_bank";

  mem test_bank_data {
    mementries = 65536;
    memwidth = 64;
    reg data {
      hdl_path = "mbp_slv_sram_bank.sram_slv_ip.InferredRam.INFERRED_SRAM.Buffer_Q";
      regwidth = 64;
      field {
        hw = r;
        sw = rw;

      } bank0[31:0] = 32'h0;
      field {
        hw = r;
        sw = rw;

      } bank1[63:32] = 32'h0;
    };
  };
  external test_bank_data bank_data[BANKS];
}; 
```

This rdl file is being include in another top level file. Now the compiler parse this just fine, but I look at the main Node structure of addrmap test, it seems like it does not recognize the reg children inside the memory. In this case, my node structure only have addrmap test, mem bank_data, but not the reg and fields inside the memory. Is this intended or am I doing something wrong?

amykyta3, 2020-05-07:
Looks like you're missing the instantiation of the reg. You're declaring the type, but not instantiating it:
```systemrdl
reg data {
   // ...
};

data data_inst[65536]; // <-- Missing this
```

hgdangkhoi, 2020-05-11:
Boy i'm dumb.... Thanks a lot...

hgdangkhoi, 2020-05-11:
Actually when I do that, I encountered error: unsupported operand type(s) for +: 'NoneType' and 'int' at 
`end_addr = last_child.raw_address_offset + last_child.total_size` in validate.py. I'm guessing that it does not set the offset of the child correctly?

hgdangkhoi, 2020-05-11:
Nevermind, it works if I do data data_inst[65536] @0x00000;.... Thank you and Sorry for your trouble, I just started to learn systemrdl.

amykyta3, 2020-05-12:
Actually I think you stumbled into a bug. Nice!

Without an explicit address allocation it *should have* automatically inferred an offset of +0x00000.
I must have missed this particular case with regs inside mem componens. Congrats on being the first one to try this :stuck_out_tongue_closed_eyes:

amykyta3, 2020-06-21:
Will be fixed in next release.

closed 2020-06-21 by amykyta3