SystemRDL/systemrdl-compiler #53: Defining regwidth which is not a power of 2

markcohen32, 2020-05-28:
Hi,

I'm looking for an option to define a regwidth which is not a power of 2. I made a search in the spec for it, but there's no sign of such an option.
Is it possible to add this option to the compiler? it would be extremely helpful.
Thanks.

amykyta3, 2020-05-29:
The semantics in the spec explicitly say that that `regwidth` shall always be a power of two (10.6.1-a).
Using anything else doesn't quite make sense since it would result in some impossible addressing implementations.

Could you describe what you are trying to accomplish? You can always use a register that contains a smaller field within it:

```
reg {
    regwidth=32;
    field {} f[23:0];
} x; // 32-bit register with only 24-usable bits
```

markcohen32, 2020-05-31:
I'm trying to build a block that will describe some events. This block isn't built from real registers, but rather of description of data structures I have in my projects. Since the data can be described in a similar manner to registers, and I need a method to collect the events from all projects, it is very convenient to use a tool that is built to do exactly that.

Regarding your example- if I define a reg that way, how bits [24:31] would look like? I mean, are they all '0'? or it is just a matter of syntax and I won't have access to them at all and the register will be treated as if it had only 24 bits?

amykyta3, 2020-05-31:
In the example I showed, the upper bits are reserved and assumed to be non-writable, as there is no logic element bound to them. In the HTML exporter, the read value is assumed to always be 0 in unused bits.

markcohen32, 2020-06-01:
Sounds good. Thanks!

closed 2020-06-21 by amykyta3