SystemRDL/systemrdl-compiler #56: Compiler not reporting error when the fields share the same location in the register

muneebullashariff, 2020-06-30:
```
    reg {
        desc = "The DATA register is used for sending and receiving data.
        Writing to the register initiates the data transmission, and the byte
        written to the register will be shifted out on the SPI output line.
        Reading the register causes the shift register receive buffer to be read,
        returning the last byte successfully received";
        
        field {
            sw=w; hw=r;
        } WDATA[7:0];
        
        field {
            sw=r; hw=w;
        } RDATA[7:0];
        
    } DATA @ 0x3;

```
As depicted above, the fields WDATA and RDATA are sharing the same location [7:0] in the register DATA.

If I run the below command the compiler **doesn't flag any error for this case**
python print_hierarchy.py atxmega_spi.rdl

Shouldn't the compiler report these kinds of errors?

amykyta3, 2020-07-01:
Although those fields occupy the same bit range, they do not collide since they use opposite access policies.
As stated in the [SystemRDL 2.0 specification](http://accellera.org/downloads/standards/systemrdl) in section 10.1-d:
> Two field instances shall not occupy overlapping bit positions within a register **unless one field is
read-only and the other field is write-only**

The example you pointed out is a demonstration of this. `WDATA` is write-only, and `RDATA` is read-only.

muneebullashariff, 2020-07-01:
@amykyta3 
Wow, that's an amazing explanation. It makes absolute sense.

Thank you!

closed 2020-07-01 by muneebullashariff