SystemRDL/systemrdl-compiler #62: UDP with multiple type

hgdangkhoi, 2020-09-16:
Hello,

I'm wondering if it is possible to create a UDP that accept multiple type? I searched the docs but can't find an example. I want to do something like:

`property assign {type=ref|boolean|string; component=signal;};`

Thanks

amykyta3, 2020-09-17:
Doesn't look like it.
Section 15 doesn't mention the ability to do so, and the BNF grammar definition in Annex B is pretty clear that it is not supported:
```
property_attribute ::=
    property_type
    | property_usage
    | property_default
    | property_constraint
property_type ::= type = property_data_type [ array_type ] ;
property_data_type ::=
    component_primary_type
    | ref
    | number
    | basic_data_type
property_usage ::= component = property_comp_types ;
property_comp_types ::= property_comp_type { | property_comp_type }
```

hgdangkhoi, 2020-09-17:
Would you be able to point me to Section 15? Also, I guess to support it first we need to modify the BNF grammar in the parser huh?

amykyta3, 2020-09-18:
I was referring to Section 15 of the SystemRDL 2.0 language spec: https://accellera.org/downloads/standards/systemrdl
See page 85.

The spec also defines the BNF grammar in Annex B, on page 107.

As this is part of the official language specification, I'm not interested in changing the compiler to violate that.

hgdangkhoi, 2020-09-21:
Thanks for pointing me to the specs. Closing this issue 👍

closed 2020-09-21 by hgdangkhoi