SystemRDL/systemrdl-compiler #63: reset value for array

soleshka, 2020-09-22:
Hello,
I have found this compiler a very useful for my register description.
The only issue i hit is how to reset an array  ? 

I ma trying to do something like : 
				reg R003_IPATOV {
					name = "R003_IPATOV";
					hdl_path   	= "";
					field {name = "field1";} field1[31:0]=0x0;

				}R003_IPATOV[8] = '{};

But i fail on ocmpilation.

Is this doable ? 

Thank you for the prompt response,
Oleg

amykyta3, 2020-09-23:
Since you already provide a reset value for `field1`, there is no need to provide a reset value for the array of registers.
In SystemRDL, reset values are only defined for the individual fields, since registers are simply a collection of fields.

soleshka, 2020-09-23:
Thank you for your prompt response. So please correct me once more, there
is no option to reset array elements with different values? I.e '{1,2,3,4..}
Thank you,
Oleg

amykyta3, 2020-09-23:
Not at the moment. The SystemRDL spec does describe the ability to do this, but this compiler does not support this yet. I may extend support for this feature in the future. See the discussion in #51 

Example: (Not supported by this compiler yet)
```
reg R003_IPATOV {
    field {} field1[31:0] = 0x0;
} R003_IPATOV[8];

R003_IPATOV[0].field1->reset = 1;
R003_IPATOV[1].field1->reset = 2;
R003_IPATOV[2].field1->reset = 3;
R003_IPATOV[3].field1->reset = 4;
```

closed 2020-12-10 by amykyta3