SystemRDL/systemrdl-compiler #65: 'include statement gives fatal error

greedysquid, 2020-09-25:
I have a .rdl file with a Verilog style include statement to pull in another .rdl file:
` _backtick_include "reg_types.rdl"`
With v1.12.0 this parses, the file is included, and all compiles.
With v1.13.0 the same files give an error:
`fatal: Unexpected text after include`

amykyta3, 2020-09-25:
Strange that my unit tests did not catch this. Would it be possible to
attach an example file?

greedysquid, 2020-09-25:
Thanks for the prompt reply!
A simple stripped down test case attached.

[test-compileRDL.zip](https://github.com/SystemRDL/systemrdl-compiler/files/5283877/test-compileRDL.zip)

amykyta3, 2020-09-25:
Thanks! Ill look at it later today. Without looking at it, my hunch is that
its related to Windows newlines vs Unix newlines.

amykyta3, 2020-09-27:
Turned out to be a regression due to the new Verilog preprocessor support added in 1.13 that broke files with Windows CRLF newlines.
Just released `v1.13.1` where it should be fixed.

Thanks again for reporting!

closed 2020-09-27 by amykyta3

greedysquid, 2020-10-13:
Apologies - I thought I had replied.
Thank you for the prompt investigation and update.
I confirm that v1.13.1 fixes this problem for me.