SystemRDL/systemrdl-compiler #66: Support for JSON serialization/deserialization

amal-khailtash, 2020-12-10:
Great work!  Very good parser.  It would be great if the elaborated parse tree could be written out as JSON and loaded from JSON.  There seems to be some circular references in the model hierarchy that makes this difficult without some code change.

amykyta3, 2020-12-17:
I don't have any plans to officially support a JSON exporter. The main reason is that there is no "standard" way to represent a register model in JSON (unlike the IP-XACT standard for XML). It would certainly be possible to export to a JSON format, but as you pointed out, the systemrdl-compiler's Python object model is much richer than what any JSON structure would be able to represent. 
Often, it is more appropriate to see if your application can directly manipulate the python object model to export what you need, rather than using an intermediate JSON format.

Since this is a request I see pretty often, I can put together an example for how to build a generic JSON exporter, and add it to the documentation here: https://systemrdl-compiler.readthedocs.io/en/latest/examples.html
Hopefully this could be used as a starting point to extend to any custom/proprietary JSON serializers you might need.

amal-khailtash, 2020-12-17:
Exporting is easier that importing.  Is there documentation on how to properly import into the object model?  In PeakRDL-ipxact I can see examples of export/import.  For export, whatever information is needed will be extracted from the model.  But for import, what exactly fields are required and what needs to be done with RDLImporter.  Documentation is missing.

amykyta3, 2021-02-15:
I have added another example that walks through building a JSON exporter here: https://systemrdl-compiler.readthedocs.io/en/latest/examples/json_exporter.html

Regarding documentation for how to build an exporter, I agree that the documentation is missing. I have to productize that process a little bit better into an actual API before I add it to the official docs. The IP-XACT importer is a good starting point, but it is a little bit ad-hoc. I'll be enhancing the [RDLImporter](https://github.com/SystemRDL/systemrdl-compiler/blob/master/systemrdl/importer.py) helper class a bit more in future releases to provide a more "official" API for this.

amal-khailtash, 2021-02-15:
Thank you.! That is great information.

amykyta3, 2021-02-21:
Just added an example on how to create your own importer: https://systemrdl-compiler.readthedocs.io/en/latest/examples/json_importer.html

This example uses some new helper functions I added to the importer class, which is now properly documented here: https://systemrdl-compiler.readthedocs.io/en/latest/api/importer.html

Note that this is not yet in published to an official release. Let me know if the tutorial and importer API accomplish what you were looking for, or if you have any recommendations.

amykyta3, 2021-02-27:
Released v1.16.0

closed 2021-02-27 by amykyta3