SystemRDL/systemrdl-compiler #67: Verilog Preprocessor Stringify/Concat

amal-khailtash, 2020-12-18:
I believe it is not possible to do this with the current preprocessor implementation:

```
`define test (r, f) \
   reg { \
     field { \
     } field``f; \
   } reg_``r; \
```

I would like to concatenate the macro parameter name to another string.

amykyta3, 2020-12-18:
Try removing the space between `test` and the opening parentheses of the argument list:
```
`define test(r, f) \
```

The Verilog preprocessor follows rules defined by the Verilog IEEE standard (from IEEE Std 1800-2012, Clause 22.5.1):
> If formal arguments are used, the list of formal argument names shall be enclosed in parentheses following
the name of the macro. The formal argument names shall be simple_identifiers, separated by commas and
optionally white space. **The left parenthesis shall follow the text macro name immediately, with no space in
between.**

amal-khailtash, 2020-12-18:
That is great. 👍

closed 2020-12-18 by amal-khailtash

amal-khailtash, 2020-12-18:
On a similar note, do you know how to escape the parameter so it does not expand?  Say, I want to do:

```
`define test(r, f) \
  r = r;
```

The first r I do not want to expand, but the second one I do.

amal-khailtash, 2020-12-18:
I guess I can choose a different name:

```
`define test(r0, f) \
  r = r0;
```

amykyta3, 2020-12-18:
Yeah you have to choose a different name. No way to escape macro tokens, even in Verilog.