SystemRDL/systemrdl-compiler #68: Deprecate the `Node.add_derived_property()` function

amykyta3, 2021-01-07:
Use of this function is not recommended since it encourages the bad practice of [modifying the SystemRDL API at runtime](https://stackoverflow.com/questions/5626193/what-is-monkey-patching). Instead, it is recommended that any extended property derivations be done in a separate function outside of the Node interface.

In the next revision make this function emit a warning notifying the user of deprecation.

After the deprecation warning is in place for a reasonable amount of time (3+ months) remove the function entirely.

amykyta3, 2021-01-26:
Deprecation warning added in 1.15.0
Function will be removed in the first release after 2020-05-01

PRWald, 2021-03-25:
I'm not seeing the deprecation warning.
<pre>
Python 3.6.5
>>> import systemrdl
>>> systemrdl.__version__
'1.17.0'
>>> def test():
...  pass
>>> systemrdl.node.RegNode.add_derived_property(test, "test")
>>>
</pre>
I see no warning message.
Also, I'm not clear about how to do things in the right way. I have a small example:
<pre>
def reg_msb_lsb(node):
    msb = None
    lsb = None
    if(isinstance(node, RegNode)):
        for f in node.fields():
            if((msb is None) or (f.msb > msb)):
                msb = f.msb
            if((lsb is None) or (f.lsb < lsb)):
                lsb = f.lsb
    return ((msb, lsb))
</pre>
which I have been adding using this command:
<pre>
RegNode.add_derived_property(reg_msb_lsb, "msb_lsb")
</pre>
Can you explain how I would do this in the future?
Thank you!

amykyta3, 2021-03-27:
Somewhat counterintuitively, deprecation warnings wont be shown by default.
If you launch python with warnings enabled, you'll see the message:
```
python3 -W all
```

The alternative usage of your function would be to simply call it directly rather than binding it to RegNode:
```python
# instead of doing this:
msb_lsb = my_reg_node.msb_lsb

# do this:
msb_lsb = reg_msb_lsb(my_reg_node)
```

amykyta3, 2021-07-02:
Removed in v1.19.0

closed 2021-07-02 by amykyta3