SystemRDL/systemrdl-compiler #70: Cannot use desc to assign value

kjeldbeg, 2021-01-13:
I have the following addrmap:

```
addrmap t_addrmap {
  reg {
    field {} iField1;
    field {} iField2;
    iField1->desc = "This is a field.";
    iField2->desc = iField1->desc;
  } reg1;
};
```
The compiler fails with:
```
example1a.rdl:6:30: fatal: 'desc' is not a known property
    iField2->desc = iField1->desc;
                             ^^^^
```
Shouldn't this be possible?

amykyta3, 2021-01-16:
In SystemRDL, not all properties are allowed to be referenced in the righthand side of an assignment expression. If you look at Table G1 in annex G, only some of the component properties are allowed to be referenced as "Ref Targets".
Furthermore, if they are referenced, they do not copy-by-value.

For example:
```
my_field_a->anded = true;
my_field_b->next = my_field_a->anded;
```
In the above example, the assignment of `anded=true` results in `my_field_a` to infer an output signal to the user that is the AND reduction of the field's value.
At the same time, `my_field_b`'s next value will inherit the AND-reduction of whatever is contained in `my_field_a` at runtime. It does not inherit the value of the property, but rather the value that the property represents.

amykyta3, 2021-01-19:
I have updated the error message to be more meaningful in this situation.
I have also added a section to the spec errata: https://systemrdl-compiler.readthedocs.io/en/latest/dev_notes/rdl_spec_errata.html#property-ref-targets
The SystemRDL spec can be pretty vague about some concepts. This one threw me off as well when I was first learning the language.

closed 2021-01-19 by amykyta3