SystemRDL/systemrdl-compiler #72: bridge address overlap

j-klier, 2021-02-02:
The example in spec section 13.5.2 shows two addrmap inside of a bridge.  SystemRDL compiler automatically assigns sequential addresses to each instance.  Such that they do not overlap.
```
ahb.ahb_credits = 0x0
ahb.ahb_stat = 0x20
axi.axi_credits = 0x80
axi.axi_stat = 0xC0
```

The description and comments in the example indicate that the registers have an overlapping address (0x0) and orthagonal addresses (0x20 and 0x40).  I had expected this as the result based on the description.
```
ahb.ahb_credits = 0x0
ahb.ahb_stat = 0x20
axi.axi_credits = 0x0
axi.axi_stat = 0x40
```

The mapping above also agrees with how UVM (uvm_reg_map) and IP-XACT (memoryMap) both allow addresses to be re-used when accessing a register from different masters.

Forcing the instances to 0x0 results in an address overlap error.
```
error: Instance 'axi' at offset +0x0:0x43 overlaps with 'ahb' at offset +0x0:0x23
    } axi @0x0;
      ^^^
fatal: Elaborate aborted due to previous errors
```

amykyta3, 2021-02-11:
Confirming this is a bug.
I need to read more to better understand how the `bridge` and `shared` properties work, and what implications they have on the compiler.

j-klier, 2021-02-22:
My understanding is that `bridge` property tells the compiler to put the `addrmap` instances into unique address spaces  such that any `addrmap` instance (and it's children) cannot collide with another `addrmap` instance within the `bridge`.  This is equivalent to having more than 1 `uvm_reg_map` in UVM or more than one `memoryMap` in IP-XACT.

The `shared` property seems similar to the `alias` keyword because it tells the compiler that a register instance exists at more than one address location.  In the case of the `shared` property, each instance must be in a different `addrmap`.  When the register is written from HW, both address locations will be updated.  When one address is written from SW, the new value can be read from the other `addrmap`.  I don't think the compiler has to do anything special here (other than pass the property through) because the linking between addresses is done in the UVM or RTL implementation.

RasmusGOlsen, 2021-03-25:
This is also my understanding. If you try to specify indirect addressing to an address map this map is seen as a different address space. Currently, I have specified the two maps independently and do an elaboration of each map, but I believe the correct way is to specify the indirectly accessible map as a bridge.

amykyta3, 2021-06-13:
Implemented what I think should be a fix.

@j-klier - Could you test and let me know if the new behavior matches your expectations?
I have not published to a release, but you can install directly form Github via:
```
python3 -m pip install git+https://github.com/SystemRDL/systemrdl-compiler.git@v1.19.0b1
```

amykyta3, 2021-07-02:
Published in v1.19.0

closed 2021-07-02 by amykyta3