SystemRDL/systemrdl-compiler #74: Compile unit and preprocessor behaves differently from SystemVerilog

j-klier, 2021-02-05:
There is a difference in behavior between SystemRDL compiler and SystemVerilog compilers with respect to how compile units are treated.  

- This page explains the SystemRDL compiler behavior
https://systemrdl-compiler.readthedocs.io/en/latest/dev_notes/multi_file_compilation.html

- SystemVerilog compilers will treat all files passed to the compiler as a common compile unit.  This specifically allows `define in one file to "leak" other files in the same compile unit.

I have a scenario where an include file is needed to provide UDP definitions.  Each file uses the UDP must `include` the file.  This allows `addrmap_a` and `addrmap_b` to be compiled individually.
**file: udp.rdl**
```
    `ifndef UDP__RDL
    `define UDP__RDL
    // property: cvr_field_vals
    //
    // UVM implementation should include field value coverage for a register
    property cvr_field_vals {type=boolean; component=reg; default=true;};

    // property: cvr_addr_map
    //
    // UVM implementation should include address map coverage for a block
    property cvr_addr_map   {type=boolean; component=addrmap|regfile; default=true;};
   `endif
```
**file: addrmap_a.rdl**

    `include "udp.rdl"

**file: addrmap_b.rdl**

    `include "udp.rdl"

When I compile them together, I can't prevent global namespace collisions because the define for `UDP__RDL` does not carry over between files.

amykyta3, 2021-02-06:
This behavior was an intentional decision in the design of this SystemRDL compiler.

Regarding your comment:
> SystemVerilog compilers will treat all files passed to the compiler as a common compile unit

This is not universally true, however this an option in most SV compilers. The SystemVerilog spec (IEEE Std 1800-2012) says the following in the introduction of clause 13.12.1:
> The exact mechanism for defining which files constitute a compilation unit is tool-specific. However,
compliant tools shall provide use models that allow both of the following cases:
a) All files on a given compilation command line make a single compilation unit (in which case the
declarations within those files are accessible following normal visibility rules throughout the entire
set of files).
b) Each file is a separate compilation unit (in which case the declarations in each compilation-unit
scope are accessible only within its corresponding file).

For the SystemRDL compiler, I have intentionally chosen to only support option B (The SystemRDL spec doesn't provide any guidance on this). I agree that treating each file as its own compilation unit does somewhat limit the usefulness of `include` directives, but doing so also removes the possibility of several much more "surprising" side effects I discuss in the dev notes. In my opinion, eliminating the possibility of these surprising namespace pollution side-effects seems more important.

Rather than including the "udp.rdl" using the preprocessor directive, recommend adding it as the first file to be compiled, as it is a dependency to all subsequent files in your project.

j-klier, 2021-02-08:
Thanks for adding the SystemVerilog Spec requirements.  I should have said that multifile compile unit (Option a) is the default behavior in most simulators.  Notice that SystemVerilog compilers must support both options.

I understand that the single file compile unit (Option b) was chosen intentionally and the reasons are well documented (thank you).  

Consider another example where a common configuration is achieved without using parameters.  This example cannot be supported without multi-file compile units.

**file: constants.rdl** 
```
    `define DATA_WIDTH 32
    `define MEM_SIZE 1024
```
**file: addrmap_a.rdl**
```
`include "constants.rdl"
addrmap addrmap_a {
    alignment = `DATA_WIDTH/8;
    mem {
        memwidth = `DATA_WIDTH;
        mementries = `MEM_SIZE;
    } external mem_a;
};
```

Please consider adding multi-file compile (Option a) as a feature request.

amykyta3, 2021-02-08:
Your include example above would work fine without any changes

closed 2021-03-12 by amykyta3