SystemRDL/systemrdl-compiler #75: Passing meta data to the code generator.

m-kru, 2021-02-09:
`systemrdl-compiler` is just front end for SystemRDL. To make any use of it one needs some back end capable of generating code.
However, it is very useful to pass meta data about component or register to such back ends. By meta data I mean any data that is not formally a part of the SystemRDL specification.  Is there any way to pass such meta data to the actual back end?
What happens with the properties which are not part of the standard?
For example
```
        field {
            desc = "Lorem ipsum";
            foo = "My meta data"

            hw=rw; we;
        } MASTER[4:4] = 0;
```
What happens with the `foo`? Is it reported as an error? 

Maybe knowledge on SystemRDL is rather limited.
Maybe there is some embedded mechanism to pass arbitrary meta data to the back end?

amykyta3, 2021-02-10:
Yep. What you are looking for are "User Defined Properties"

To take your example, you can declare the existence of a UDP directly in RDL:
```
// Delcare a UDP called "foo" that shall be a string type, and is only valid for field components
property foo {
    type = string;
    component = field;
};

field {
    desc = "Lorem ipsum";
    foo = "My meta data"
    hw=rw; we;
} MASTER[4:4] = 0;
```

Similarly, you can use the compiler API to declare your own builtin UDP: https://systemrdl-compiler.readthedocs.io/en/latest/api/compiler.html#systemrdl.RDLCompiler.define_udp

amykyta3, 2021-02-10:
To clarify on your original question, yes - if the compiler encounters a property assignment that is not part of the official specification, or one that has not been declared as a user property, then the compiler will flag it as an error.

That said, definitely check out the spec - specifically Table G1 which has a summary of all the properties. The scope that the builtin properties covers is already quite broad, so its possible the spec already covers what you need. I also summarized all the properties here in this page: https://systemrdl-compiler.readthedocs.io/en/latest/property_reference.html

closed 2021-02-10 by m-kru