SystemRDL/systemrdl-compiler #77: ref not allowed as parameter type

sorenthe, 2021-03-09:
I have noticed that the compiler does not allow `ref `as a parameter type as in this example `#(ref p)`:

```
addrmap a_map #(ref p) {
  reg {
    field {} a_field;
  } a_reg;
};

```
Looking in the Standard (jan 2018) section 5.1.1.1 and Table 7 this seems to be allowed.

amykyta3, 2021-03-12:
Not quite.

Clause 5.1.1.2-e explicitly says that parameters cannot be references:
> e)  Component instance references shall not be used as parameter values (either directly or as part of an
aggregate type).

Furthermore, the grammar in Annex B6 & B14 does not include the `ref` or other component keywords in the parameter type:
```
param_def_elem ::= data_type id [ array_type ] [ = constant_expression ]
...
data_type ::=
    basic_data_type
    | accesstype
    | addressingtype
    | onreadtype
    | onwritetype
basic_data_type ::=
    simple_type [ signing ]
    | string
    | boolean
    | id
simple_type ::= integer_type
```

The link to Table 7 should probably not be taken too literally. This looks like just another of [many minor inconsistencies in the spec](https://systemrdl-compiler.readthedocs.io/en/latest/dev_notes/rdl_spec_errata.html).

sorenthe, 2021-03-12:
Clause 5.1.1.2-e decides it of course. Pity though, I could have used the feature.

closed 2021-03-20 by amykyta3