SystemRDL/systemrdl-compiler #78: Size of mem instantiation is wrong

sorenthe, 2021-03-11:
When instantiating a memory I unexpectedly get this error.

```
mem_space.rdl:11:16: error: Address space occupied by registers (0x250) exceeds size of mem 'm' (0x14A)
  external MEM m;
               ^
fatal: Elaborate aborted due to previous errors

```

This is the RDL I used:

```
addrmap TOP {
  mem MEM {
    mementries = 165;
    memwidth   = 22;

    reg REG {
      regwidth = 32;
      field {} f[22];
    } rg[148];
  };

  external MEM m;
};

```
I think the size of the memory m is calculated wrong. The value calculated is 0x14a = 350 = 165 * 2. It should have been 165*(2^2) rounding the size of a memory word up to the nearest power of 2 bytes.

A naïve fix is:
```

#===============================================================================
class MemNode(AddressableNode):

    @property
    def size(self) -> int:
#        entry_size = self.get_property('memwidth') // 8
        entry_size = 1
        while entry_size < (self.get_property('memwidth') + 7) // 8:
            entry_size = entry_size * 2;
        num_entries = self.get_property('mementries')
        return entry_size * num_entries

```

amykyta3, 2021-03-12:
Thanks for reporting!
Confirm that definitely looks wrong.

Should be:
```python
entry_size = helpers.roundup_pow2(self.get_property('memwidth')) // 8
```

sorenthe, 2021-03-12:
Will this work for memwidth  <= 4?

amykyta3, 2021-03-13:
Nope, but it will now :smile:

amykyta3, 2021-03-20:
Published fix in v1.17.0

closed 2021-03-20 by amykyta3