SystemRDL/systemrdl-compiler #79: Namespace when compiling multiple files

Legendary2020, 2021-03-18:
Hello,

I have a question about compiling multiple files. Let's say I define multiple UDP properties inside a file called "udp.rdl", here are my inputs:

top.rdl:
```
`include "udp.rdl"
`include "a1.rdl"
`include "a2.rdl"

addrmap top {
        a1 a1 @0x0;
        a2 a2 @0x100;
};
```

a1.rdl:
```
`include "udp.rdl"

addrmap a1{
           ....
}
```

a2.rdl:
```
`include "udp.rdl"

addrmap a2{
           ....
}
```

First approach: now, if I compile the addrmap top, everything is fine, although because I have the `include "udp.rdl"` in multiple files, shouldn't this cause a namespace conflict and trigger "Multiple declarations of user-defined property ....." ?

Second approach: if I remove these 2 lines `include "a1.rdl"` and `include "a2.rdl"` and compile 3 files together, I would get the UDP name collisions above. Why there is a difference between these 2 approach?

We are using multiple RDL files and compile everything from the top, so using the `include works nicely, but is it recommend to compile all the files like the second approach?

amykyta3, 2021-03-19:
What does `udp.rdl` look like? Are you using include guards like this?
```
`ifndef UDP_RDL
`define UDP_RDL

property my_property {
    ...
};

`endif
```

Just like the type namespace (addrmap, reg, field, etc declarations), UDPs can only be declared once in the root namespace per project. As you mentioned,  re-declarations will throw an error. This can occur if compiling a single file that includes `udp.rdl` multiple times (directly or indirectly), or if encountered during separate file compiles. I'm assuming that your `udp.rdl` file uses include guards (ifndef+define) which is why in your "First approach", you don't see any error - the include guards would prevent subsequent inclusions of `udp.rdl` from re-defining the UDPs. If compiling the files separately, they are treated as distinct compilation units, so `define` macros do not carry-over.

See my description on how I handle namespaces & compilation units for more details: https://systemrdl-compiler.readthedocs.io/en/latest/dev_notes/multi_file_compilation.html

Personally, I prefer not to use the `include` directive at all (or very sparingly). Instead, compile each RDL file separately, including `udp.rdl`. Simply be sure to compile `udp.rdl` first so that you get your project-specific UDPs loaded into the root namespace ahead of any other RDL files.

Legendary2020, 2021-03-19:
You're correct that I'm using the include guards in my `udp.rdl`. I checked the code but I can't find which part/functions that bypass the redefining `udp.rdl` if using the include guard, can you help point me to where that is? 

Thank you!

amykyta3, 2021-03-19:
The include guards are what cause the include to get bypassed.
Lets take a simpler example of `top.rdl` as follows:

```verilog
// One
`include "udp.rdl"

// Two
`include "udp.rdl"
```

When the preprocessor expands the includes, it will flatten the text into:
```verilog
// One
`ifndef UDP_RDL
`define UDP_RDL
property my_property {
    ...
};
`endif

// Two
`ifndef UDP_RDL
`define UDP_RDL
property my_property {
    ...
};
```

The first time, `UDP_RDL` is not defined, so the contents of the `ifndef` get processed.
The second time, `UDP_RDL` has already been defined, so the contents of the second `ifndef` are skipped.
So the final preprocessed result is:
```verilog
// One
property my_property {
    ...
};

// Two
```

Legendary2020, 2021-03-19:
That is very helpful. However, if we compile multiple files that have `include "udp.rdl"`, why don't the include guard in `udp.rdl` trigger and prevent the contents of the file being redefined again? Is it because each file is its own compilation unit?

```
// File 1
`include "udp.rdl"

```

```
// File 2
`include "udp.rdl"
```

We compile 2 files. In this case, each file is its own compilation unit, the contents of the `udp.rdl` `UDP_RDL` is defined in compilation unit of file 1. In file 2, even with the include guard, because in the compilation unit of file 2 does not have `UDP_RDL` yet, it is being defined again. However, because the UDP namespace is global, this causes a name conflict by redefining the property. Is this correct?

Legendary2020, 2021-03-19:
I released there is another issue that the compiler does not detect when compiling multiple file. Let's change my first example a little bit:
top.rdl:
```

`include "udp.rdl"
`include "a1.rdl"
`include "a2.rdl"

addrmap top {
        a1 a1 @0x0;
        a2 a2 @0x100;
};
```

a1.rdl:
```
`include "udp.rdl"
`include "a2.rdl" //addrmap a2 now exists in the global type namespace
addrmap a1{
           ....
}
```

a2.rdl:

```
`include "udp.rdl"

addrmap a2{ // expect a redeclaration error because a2 is in global namespace and already defined
           ....
}
```

In the example above, compiling only the top, shouldn't error be thrown for `addrmap a2` because of the `include a2.rdl` in `a1.rdl`?

amykyta3, 2021-03-20:
> if we compile multiple files that have include "udp.rdl", why don't the include guard in udp.rdl trigger and prevent the contents of the file being redefined again? Is it because each file is its own compilation unit?

Yes exactly. Each call to `RDLCompiler.compile_file()` is handled in its own compilation unit. Macros are discarded at the end of each compilation unit, so if compiling separately, the include guards will be unable to prevent re-inclusion.

> In the example above, compiling only the top, shouldn't error be thrown for addrmap a2 because of the include a2.rdl in a1.rdl?

Yes, if the files were compiled exactly as you hve them written, then I'd expect there to be an error thrown. But I assume you have `ifndef/define` include guards in `a2.rdl` as well?

Legendary2020, 2021-03-20:
> But I assume you have ifndef/define include guards in a2.rdl as well?
No, the a2.rdl is normal rdl file with an addrmap definition. The ifndef/define is only used in `udp.rdl`. I don't know why the compiler did not catch this error.

amykyta3, 2021-03-20:
I cannot reproduce the issue you are seeing.
Can you confirm what version of the systemrdl-compiler you are using?
```bash
$ python3.7 -c "from systemrdl import __version__; print(__version__)"
1.17.0
```
Also, is this Windows/Linux/Mac? Python version?

I ran with the following files, exactly as follows:

**top.rdl**
```
`include "udp.rdl"
`include "a1.rdl"
`include "a2.rdl"

addrmap top {
    a1 a1 @0x0;
    a2 a2 @0x100;
};
```

**udp.rdl**
```
`ifndef UDP_RDL
`define UDP_RDL
property foo {
    component = all; type = boolean;
};
`endif
```

**a1.rdl**
```
`include "udp.rdl"
`include "a2.rdl"
addrmap a1{
    reg {
        field {} xyz;
    } a1_reg;
};
```

**a2.rdl**
```
`include "udp.rdl"

addrmap a2{
    reg {
        field {} xyz;
    } a2_reg;
};
```

When I run it through my "[print_hierarchy.py](https://github.com/SystemRDL/systemrdl-compiler/blob/master/examples/print_hierarchy.py)" example, I see the re-declaration error as expected:
```bash
$ ./examples/print_hierarchy.py tmp/top.rdl 
tmp/a2.rdl:3:9: fatal: Multiple declarations of type 'a2'
addrmap a2{
```

Legendary2020, 2021-03-22:
I think you are correct, I tested again this morning using a clean workspace and got the re-declaration error for my example above. Thanks a lot for helping me understand this issue.

closed 2021-03-22 by Legendary2020