SystemRDL/systemrdl-compiler #80: How to get node's path with array dimension size

RasmusGOlsen, 2021-03-19:
I can see the node.get_path has an array_suffux argument and an empty_array_suffix argument. It seems like the array_suffix argument is used when looping over nodes with unroll=True and the empty_array_suffix is used when looping over nodes with unroll=False.
The default argument for array_suffix is "[{index:d}]" which gives me the array index for each array dimension. Can I do something similar with the empty_array_suffix, but with the array size instead, like this "[{size:d}]"?

amykyta3, 2021-03-19:
Not currently, but thats a really good idea!

Ill add that feature. Ill use "dim" instead of "size" to avoid confusion
with Node.size

amykyta3, 2021-03-20:
Added in v1.17.0
See: https://systemrdl-compiler.readthedocs.io/en/latest/api/node.html#systemrdl.node.Node.get_path

closed 2021-03-20 by amykyta3