SystemRDL/systemrdl-compiler #81: Unroll node that is an array

RasmusGOlsen, 2021-03-22:
How do I unroll/loop through a node that is an array? I know you can unroll when you loop through the descendants or children of node, but what if I want to loop through the array of the node itself?

amykyta3, 2021-03-23:
The `Node` interface doesn't have a built-in method to do that currently.
I'll add that as a feature request since I can see how that would be useful.

In the meantime, you can use this function to unroll any node that is an array. This should be pretty future-proof too:
```python
from typing import Iterator
import itertools
from systemrdl.node import Node, AddressableNode

def unrolled(node: Node) -> Iterator[Node]:
    cls = type(node)
    if isinstance(node, AddressableNode) and node.is_array:
        # Is an array. Yeild a Node object for each instance
        range_list = [range(n) for n in node.array_dimensions]
        for idxs in itertools.product(*range_list):
            N = cls(node.inst, node.env, node.parent)
            N.current_idx = idxs
            yield N
    else:
        # not an array. Nothing to unroll
        yield cls(node.inst, node.env, node.parent)
```

RasmusGOlsen, 2021-03-24:
It would also be useful to be able to unroll a single dimension.

amykyta3, 2021-08-31:
Implemented in v1.20.0

closed 2021-08-31 by amykyta3