SystemRDL/systemrdl-compiler #83: How to use a reference with swwel access property

PRWald, 2021-03-31:
Here is a sample RDL file in which I am trying to use the **enable** bit to lock the other registers, so that if **enable** is low I can write to the other registers, but if **enable** is high, I cannot.
<pre>
addrmap foo_am {
    regfile foo_rf_t {
        default regwidth=8;
        reg {
            field {
                sw=rw; hw=r;
            } enable[0:0] = 0;
        } ctrl_reg;
        reg {
            field {
                sw=rw; hw=r;
            } foo_f[7:0] = 0x00;
        } foo_reg_1;
        reg {
            field {
                sw=rw; hw=r;
            } foo_f[7:0] = 0x00;
        } foo_reg_2;
    } foo_rf;

    foo_rf.foo_reg_1.foo_f->swwel = foo_rf.ctrl_reg.enable;
    foo_rf.foo_reg_2.foo_f->swwel = foo_rf.ctrl_reg.enable;
};
</pre>
The following script will then go through all the fields and print the WE or that it has no WE.
<pre>
#!/usr/bin/env python3

from systemrdl import RDLCompiler, RDLListener, RDLWalker

class listener(RDLListener):
    def __init__(self, root):
        self.root = root

    def enter_Field(self, node):
        path = node.get_path()
        print('Field %s' % path, end='')
        if('swwel' in node.inst.properties):
            we = node.inst.properties['swwel']
            print(' is write enabled by %s' % we.get_path())
        else:
            print(' has no write enable')

if(__name__ == '__main__'):
    rdlc = RDLCompiler()
    rdlc.compile_file('sample.rdl')
    root = rdlc.elaborate()
    walker = RDLWalker()
    listener = listener(root)
    walker.walk(root, listener)
</pre>
(BTW, I'm using 1.17.0 and Python 3.5.1.)
I now know that **we** has no .get_path() method. Using the Python debugger I see:
<pre>
(Pdb) we
&lt;systemrdl.rdltypes.ComponentRef object at 0x7f4698c4bdd8&gt;
(Pdb) we.__dict__
{'ref_elements': [('foo_rf', [], <systemrdl.source_ref.SegmentedSourceRef object at 0x7f4698c4b898>), ('ctrl_reg', [], &lt;systemrdl.source_ref.SegmentedSourceRef object at 0x7f4698c4b8d0&gt;), ('enable', [], &lt;systemrdl.source_ref.SegmentedSourceRef object at 0x7f4698c4b940&gt;)], 'ref_root': &lt;Addrmap foo_am at 0x7f4698cb5780&gt;}
</pre>
I'm just not sure what to do with this type of object and what my options are. I'm trying to write a script to generate RTL. I guess all the information I need can be gotten to from this object, but it seems like a lot of effort to extract it, so I figured I'm just missing something, not thinking about it right, or completely off in my understanding of RDL, software access properties, etc. Any advice would be very much appreciated.

amykyta3, 2021-04-01:
I would definitely avoid accessing `node.inst.properties` directly. The `properties` dictionary is not in the official documentation and is used for internal bookkeeping. I cant make any guarantees about it in case I change how structures are stored internally. Also as you've noticed, component references are stored there in an internal encoded form which isn't really useful to you.

Instead, **always** use the `node.get_property()` function when querying. This function will provide you a Node object if there are references. It will also automatically return the inferred default of the `swwel` property if it was not explicitly assigned by the user. SystemRDL has a bunch of subtle implied behaviors in properties that this will handle for you automatically.
See docs: https://systemrdl-compiler.readthedocs.io/en/latest/api/node.html#systemrdl.node.Node.get_property

So to transcribe your example a bit:
```python
from systemrdl.node import Node

def enter_Field(self, node):
    path = node.get_path()
    print('Field %s' % path, end='')
    
    swwel = node.get_property('swwel')
    if isinstance(swwel, Node):
        # User assigned a component reference. Property is a Node object
        print(' is write enabled by reference to %s' % swwel.get_path())
    elif swwel is True:
        # User assigned "true". Your tool should create an implicit signal that controls the write-enable
        print(' is write enabled by an implicit signal (tool-defined)')
    else: # swwel is False
        print(' has no write enable')
```

In case you REALLY want to know if the user assigned it, you can set a default value to `get_property()`:
```python
swwel = node.get_property('swwel', default=None)
if swwel is None:
    print('swwel was not assigned explicitly')
```
But the situations where you'd need this are rare, and should be avoided unless you have a _really_ good reason.

PRWald, 2021-04-01:
Perfect! Thank you so much, Alex.

closed 2021-04-04 by amykyta3