SystemRDL/systemrdl-compiler #85: Obtaining the line number

kjeldbeg, 2021-05-12:
If I discover a problem with a node (e.g. a property is not set) I would like to generate an error message with the line number from the RDL file. Is that information available?

amykyta3, 2021-05-13:
Since you simply want to print messages to the user, I recommend using the existing message handler. This will print messages in a format that is consistent with the rest of the compiler, and also include source context. I assume this will do what you need implicitly.

Here is an example that uses the [atxmega_spi.rdl](https://github.com/SystemRDL/systemrdl-compiler/blob/master/examples/atxmega_spi.rdl#L64-L66) demo file:
```python
from systemrdl import RDLCompiler
rdlc = RDLCompiler()
rdlc.compile_file("atxmega_spi.rdl")
root = rdlc.elaborate()

# Print messages without any context
rdlc.msg.warning("Warning Message")
rdlc.msg.error("Error Message")

# Print messages with source context
data_if_field = root.top.find_by_path("STATUS.IF")
rdlc.msg.warning(
    "Field definition is here",
    data_if_field.inst.def_src_ref
)
rdlc.msg.warning(
    "Field instance is here",
    data_if_field.inst.inst_src_ref
)
rdlc.msg.warning(
    "Assignment of the 'hw' property is here",
    data_if_field.inst.property_src_ref['hw']
)
```
Running the above script produces this output:
[image]

I'll add the `RDLCompiler.msg` attribute to the documentation. That really ought to be part of the public API.

amykyta3, 2021-05-13:
Updated docs:
* https://systemrdl-compiler.readthedocs.io/en/latest/api/compiler.html#systemrdl.RDLCompiler.msg
* https://systemrdl-compiler.readthedocs.io/en/latest/api/messages.html#systemrdl.messages.MessageHandler
* https://systemrdl-compiler.readthedocs.io/en/latest/api/component.html#systemrdl.component.Component.inst_src_ref

kjeldbeg, 2021-05-17:
Just what I needed. Thank you very much.

closed 2021-05-17 by kjeldbeg