SystemRDL/systemrdl-compiler #100: Instance reference in struct

j-klier, 2021-07-09:
I'm trying to create a struct with an instance reference as one of the members.  Table 7 says to use the keyword `ref`

```
struct submap_t {
    ref child_map;
    bit offset;
} ;
```

This results in a compile error.
```
extraneous input 'ref' expecting {'}', 'boolean', 'bit', 'longint', 'string', 'accesstype', 'addressingtype', 'onreadtype', 'onwritetype', 'addrmap', 'regfile', 'reg', 'field', 'mem', 'signal', ID}
    ref child_map;
    ^^^
```

Is this supported?

amykyta3, 2021-07-10:
The `ref` keyword is not allowed in this situation. Instead, try one of the specific component type keywords like `reg`, `field`, etc...
That should work.

If you look at the formal grammar definition in annex B.4, struct elements are defined as follows:

```
struct_elem ::= struct_type id [ array_type ]
struct_type ::= data_type | component_type
...
component_type ::= component_primary_type | signal
component_primary_type ::= addrmap | regfile | reg | field | mem
```

I've found that the RDL spec is *extremely* inconsistent when it comes to the `ref` keyword. They prominently display it in Table 7,  but the only place the formal BNF grammar actually allows it is in user defined property definitions. Nowhere else!

j-klier, 2021-07-13:
I've confirmed this works as expected.  Thanks!
```
struct submap_t {
    addrmap child_map;
    bit offset;
};
```

closed 2021-07-13 by j-klier