SystemRDL/systemrdl-compiler #102: User Defined Property cannot reference a signal

j-klier, 2021-07-15:
Table 30 shows that `ref` is a reference to a component instance.  Where "component" comes from Section B.3
```
component_type ::=
    component_primary_type
  | signal
component_primary_type ::= addrmap | regfile | reg | field | mem
```

Does `Signal` need to be added to the `valid_types` in `UDPVisitor.py`?

https://github.com/SystemRDL/systemrdl-compiler/blob/master/systemrdl/core/UDPVisitor.py
```
            valid_types = [
                comp.Field,
                comp.Reg,
                comp.Regfile,
                comp.Addrmap,
                comp.Mem,
                comp.Signal
            ] # type: List[Type[Union[int, str, bool, rdltypes.BuiltinEnum, rdltypes.UserEnum, rdltypes.UserStruct, comp.Component]]]

 ```

amykyta3, 2021-07-16:
See my errata page on this topic: https://systemrdl-compiler.readthedocs.io/en/latest/dev_notes/rdl_spec_errata.html#user-defined-property-s-type-attribute-can-not-be-signal

I agree this is an odd limitation, but this is what the official RDL spec specifies. It would be good to get clarification from the Accellera committee members why this limitation was imposed.

j-klier, 2021-07-16:
Thanks for documenting this.  I understand your choice to omit signal from UDP references.  I agree that `type=signal` is not allowed.  However, I don't agree that the spec is clear on how to interpret the `ref` keyword.  

Table 31 says:
> addrmap, regfile, reg, mem, field = A specialization of the ref keyword to instances of given type.

Notice that it doesn't say that these are the only possible specializations. 

Table 30 says:
>      `ref` = "A reference to a component instance" 

A literal interpretation suggests that `type=ref` this is allowed to reference a signal since signals are component instances.

amykyta3, 2021-07-16:
Yeah. The definition of the ref keyword has a lot of inconsistencies. This
isn't the only one. Theres also this:
https://systemrdl-compiler.readthedocs.io/en/latest/dev_notes/rdl_spec_errata.html#inconsistent-definition-of-the-ref-type-keyword

closed 2021-08-13 by amykyta3