SystemRDL/systemrdl-compiler #103: Negative bit numbers for field

kjeldbeg, 2021-07-28:
I have this addrmap:
```
addrmap mymap {
  reg {
    field {
    } myfld1[14:16];
    field {
    } myfld2[20];
  } myreg;
};
```

When I print high/msb and low/lsb for the fields I get:

```
myfld2 13:-6 13:-6
myfld1 16:14 16:14
```
It seems the problem occurs because I specify 14:16 instead of 16:14.

amykyta3, 2021-07-29:
Good find. This was a bug in validation. Next version will flag this as a compilation error as follows:
```
bug103.rdl:6:7: fatal: Field 'myfld2' of width 20 infers bit range [-6:13] which extends below bit 0
    } myfld2[20];
      ^^^^^^
```

The way you specify bit-order of fields also determines how subsequent fields are packed. For example:
```
reg {
    field {} a[12:8]; // implies this reg is "lsb0" order
    field {} b[4]; // inferred as b[16:13]
} r1;

reg {
    field {} c[8:12]; // implies this reg is "msb0" order
    field {} d[4]; // inferred as b[7:4]
} r2;
```

This behavior is defined in the spec under clause 9.2-e:
> In the mode **msb0**, unless bit allocation is explicitly defined, fields shall be positioned sequentially
> in the order they are instantiated in a register, starting at bit `regwidth-1` with no padding between
> fields. (Each subsequent field’s least significant bit (LSB) shall be made equal to one (1) less than
> the most significant bit (MSB) of the previous field.)

and in 10.7-b:
> `field_type field_instance [low:high]`
> implies the use of **msb0** ordering

In your example, the 20-bit field `myfld2` was correctly packed below `myfld1`, but validation did not catch that it was out of bounds.

amykyta3, 2021-08-31:
Implemented in v1.20.0

closed 2021-08-31 by amykyta3