SystemRDL/systemrdl-compiler #104: Calculation of memory component address space wrong?

hekner, 2021-07-29:
I believe the address calculation for memories is done wrongly, ie. it is done as for registers which doesn't make sense physically. The standard is very vague here - maybe there are different interpretations?

The RDL example below is a trivial example of a 32-bit physical bus system, where there is a memory attached which is only 13 bits wide and 4 entries deep. This means that the physical 4deep x 13wide memory is connected to the bus system taking up 4 32-bit words address space (0x10 total address space), and within each of those 4x 32-bit words, the memory is connected to the 13 lowest bits in each word.

Even so, the tool calculates ((wrongly in my opinion) that the memory only takes up 2 words (0x8 address space), which is physically impossible. A 4-entry memory (with any width from 1-32 bits in this case) will *always* take up 4 words (0x10 address space in this case).

If you believe that the address/size calculation done by the tool today is correct, can you point to the location in the standard where this is described?

default regwidth    = 32;
default accesswidth = 32;

addrmap memexample {
  external mem {
    memwidth   = 13;
    mementries = 4;
  } mymem;

  reg { field {} r8[8] = 0; } reg8;
};

amykyta3, 2021-07-30:
The default assignments of `regwidth` and `accesswidth` properties only apply for `reg` components. These properties are not valid for `mem` components so they do not have any effect (see 5.1.3.2).

The SystemRDL spec does not provide any explicit guidance on how a memory is mapped to an address space, so the only remaining interpretation is to allocate bytes by rounding up the `memwidth` to the next power-of-two bytes as is implied by 11.2-e.
The behavior of inferred address allocation is also not defined very clearly in the spec. For this reason the compiler has an [optional warning you can enable](https://systemrdl-compiler.readthedocs.io/en/latest/api/messages.html#systemrdl.warnings.IMPLICIT_ADDR) that encourages users to explicitly assign addresses.

hekner, 2021-07-30:
Thanks for the detailed answer.

closed 2021-07-31 by amykyta3