SystemRDL/systemrdl-compiler #107: Use of "wel" property generates error if field is not writable from SW

hekner, 2021-09-04:
addrmap wel_example {
  reg { field {sw=r; hw=w; wel;} data[8]; } my_reg;
};

Gives the following error:

error_wel.rdl:2:28: error: Use of 'wel' property on field 'data' that does not implement storage does not make sense
  reg { field {sw=r; hw=w; wel;} data[8]; } my_reg;
                           ^^^

This seems to be a mistake. "wel" indicates there is a load enable (active low) on the HW write port - and has nothing to do with the software access.

Ie. the code implements a register writable by HW (with storage of course) under the control of a load_enable signal, which can then be read by SW.

So I believe this code is fully legal and should not generate an error. Same of course applies if "wel" is replaced with "we".

BR, Hartvig

amykyta3, 2021-09-04:
This is intentional.
Although I agree that the `we` and `wel` properties are not directly related to the software access mode used, the combination of `sw` and `hw` properties have clearly defined effects on whether the field infers a storage element or not. See section 9.4.1, Table 12 in the RDL 2.0 spec:
[image]

In your situation, `sw=r; hw=w;` infers a field which is directly assigned a combinational value from the hardware interface - specifically, the register block itself does *not* infer a storage element in its implementation.

The reason for the error you are seeing is that use of a write-enable strobe (`we` or `wel`) is meaningless in this situation. This is because there is no storage element (DFF) that can even be write-enabled. By definition, a combinational wire/bus input is always write-enabled - it cannot be disabled.

If your intent was to have:
* a software read-only field
* that infers a storage element
* that is writable by hardware
* and is dynamically able to be write-enabled by the hardware
... you can do the following:
```
field {
    sw=r; hw=rw; // Infers a software read-only field. hw=rw enables a storage element
    wel;
} data[8];
```
This has the side effect of also exposing the storage element output back to hardware (Q pin of the D-flip-flop).

hekner, 2021-09-04:
Thanks for the detailed clarification, I hadn't noticed this is table 12.
I will add the hw "r" port (although unused) to force the presence of the storage element.

You may close the case.

closed 2021-09-05 by amykyta3