SystemRDL/systemrdl-compiler #116: Nested regfile definition

alibabashack, 2021-11-26:
From my interpretation of the SystemRDL 2.0 standard, the following nested definition of regfile should be supported:
```
regfile foo {
regfile bar {

    reg {
        field {
        } someField;
    } MyReg;

};
};

addrmap example_soc {
    foo myPeripheralA @ 0xA000;
};
```
When running PeakRDL-html on this file I get
```
myreg2.rdl:14:9: error: Register file 'myPeripheralA' must contain at least one reg or regfile.
    foo myPeripheralA @ 0xA000;
        ^^^^^^^^^^^^^
fatal: Elaborate aborted due to previous errors
```
I also thought of accessing the nested regfile bar using this syntax:
```
addrmap example_soc {
    foo.bar myPeripheralA @ 0xA000;
};
```
It results in:
```
myreg2.rdl:14:13: error: no viable alternative at input 'foo.barmyPeripheralA'
    foo.bar myPeripheralA @ 0xA000;
            ^^^^^^^^^^^^^
fatal: Parse aborted due to previous errors
```
I am unsure if this is a result of my incompetence or a bug.
In any way, thanks for this awesome project.

alibabashack, 2021-11-26:
I solved the problem. Apparently, when the outer regfile is instantiated, this will not implicitly instantiate the inner regfile definition. Instead, such an instantiation needs to be explicit:
```
regfile foo_t {
regfile bar_t {

    reg {
        field {
        } someField;
    } MyReg;

} bar;
};

addrmap example_soc {
    foo_t myPeripheralA @ 0xA000;
};
```
This makes sense, since it allows to have multiple instantiations of bar_t within foo_t.

I will leave this issue open, because I did not find a test case for such nesting. Maybe one should be added? Otherwise, please feel free to close this.

amykyta3, 2021-11-26:
Correct. In your first example, you're creating a *definition* of `bar`, but not actually instantiating it.

If you don't plan to instantiate something multiple times, you can omit the type name and use an anonymous declaration:
```
regfile foo_t {
    regfile { // <--- Anonymous declaration
        reg {
            field {
            } someField;
        } MyReg;
    } bar;
};

addrmap example_soc {
    foo_t myPeripheralA @ 0xA000;
};
```

closed 2021-11-26 by amykyta3