SystemRDL/systemrdl-compiler #118: INTR not supported at reg definition

deanMaker, 2022-01-10:
While identifying a field with an intr property is supported, doing the same at reg level throws an error.  From table 23 of the LRM, this should be allowed.  
Example snippet:
```
reg {
        name = "SPI control register 2 (SPI_CR2)";
        intr;    // <- throws error
        field {
            hw   = r;
            sw   = rw;
            intr;          
            name = "RX buffer not empty interrupt enable";
            encode = RXNEIE_enum;
        } RXNEIE = 0;
    } CTRL2 @ 0x4;
```

amykyta3, 2022-01-11:
Not quite.
Although you are correct that Table 23 lists `intr` as a property of `reg`, it is not something that you can assign. Note how the "type" column shows "N/A".
Section 10.8 clarifies this in much more detail:

> Any register that contains an interrupt field has two implied properties: `intr` and `halt`. **These properties are outputs of the register**. The `intr` register property represents the inclusive OR of all the interrupt bits in a register after any field enable and/or field mask logic has been applied.

What this means is that for registers, the `intr` property can only be accessed as a righthand side of an assignment.
For example:
```
some_property = my_reg->intr;
```

deanMaker, 2022-01-11:
Thanks for the clarification.

closed 2022-01-11 by deanMaker