SystemRDL/systemrdl-compiler #119: Importer "name" validation is too strict.

jasonpjacobs, 2022-01-11:
The '_instantiate' method of the RDLImporter class performs some basic validation of instance names, using the following approach:

```
if not re.fullmatch(r"\w+", inst_name):
            raise ValueError("Instance name has invalid characters")
```

This seams to be a bit too strict.  The IP-XACT spec (1685-2014, Section D.7) has the following description of allowed characters in names:

> The Name type defines a series of any characters, excluding embedded whitespace. It needs to begin with a
> letter, colon (:), or underscore (_). A Name shall contain only letters, numbers, and the colon (:),
> underscore (_), dash (-), and dot (.) characters. Any leading or trailing spaces are removed.

This seems like a straightforward fix that I may take on if it isn't already planned.

amykyta3, 2022-01-12:
I assume you're getting this exception while using the [PeakRDL-ipxact](https://github.com/SystemRDL/PeakRDL-ipxact) importer? Or is this your own IP-XACT tool?

Although you are correct that IP-XACT supports a broader range of characters, remember that the SystemRDL language only allows alphanumeric identifiers. The intent of the importer API is to help translate foreign input structures such as IP-XACT into an internal object model that aligns with SystemRDL semantics. For that reason, I don't think it would be appropriate to loosen the rules of the importer, especially since downstream tools implicitly have the expectation that the register model will have been validated according to SystemRDL rules. Usually SystemRDL is the more expressive language so stuff like this isn't an issue. I actually didn't notice this about IP-XACT until you pointed it out (thanks!!). Would you be able to share some register/field names that use these extra characters? I'm mostly curious about how ip-xact is used out in the wild.

If you're getting this error from the PeakRDL-ipxact tool, I think it would be more appropriate to have the importer automatically sanitize identifiers into RDL-compatible ones. For example: transform `my-register.name:3` into `my_register_name_3`.

jasonpjacobs, 2022-01-12:
Hi.  Thanks for the prompt response!

Yes, you are correct in that the error I've seen was an IP-XACT import error.  And I agree: The SystemRDL importer should not accept names/identifiers not allowed by SystemRDL.  Also... I didn't realize that SystemRDL rules for identifiers was more strict than IP-XACT.  I'll create an issue in the PeakRDL-ipxact project.

closed 2022-01-12 by jasonpjacobs

jasonpjacobs, 2022-01-12:
And to answer your question about IP-XACT sources from the field, the types of names that violate SystemRDL naming conventions are mostly names that include periods (e.g. "some_ip.some_signal").  I haven't come across colons, semi-colons, or slashes.