SystemRDL/systemrdl-compiler #123: alias reg does not inherit the external property

j-klier, 2022-03-02:
This is a minor bug that I noticed.  Alias registers do not inherit the external property from the primary register as required in section 10.5.1.f 
> If the alias instance type not specified, it uses the primary register instance type.

I think this can be fixed by resolving the external property during elaboration (elaborate.py).
```
    def enter_Reg(self, node: RegNode) -> None:
        if self.coerce_external_to is not None:
            # Is nested inside regfile that is coercing to a particular inst type
            node.inst.external = self.coerce_external_to
        elif node.inst.external is None:
            if node.inst.is_alias:
                node.inst.external = node.inst.alias_primary_inst.external
            else:
                node.inst.external = False
```

amykyta3, 2022-03-04:
Thanks!
Alias registers are still something I need to spend more time implementing in general. Currently, I don't think the compiler checks any of the clauses listed in 10.5.1!

Your suggested change looks like the right place to implement 10.5.1-f so I'll include it in the next update, hopefully along with other alias register support.

amykyta3, 2022-03-17:
Fixed in v1.23.0.

I also added a bunch of tools to `RegNode` and `FieldNode` that should make it easier to manipulate aliases. Hopefully you will find these useful. See the release notes: https://github.com/SystemRDL/systemrdl-compiler/releases/tag/v1.23.0

closed 2022-03-17 by amykyta3