SystemRDL/systemrdl-compiler #124: Setting register default offset

markcohen32, 2022-03-21:
Hi,

When instantiating two or more registers one after another without setting the address explicitly , the compiler arranges their address according to their width, for example, two registers with 32 bits width:

register_1 register_1_inst;
register_2 regisetr_2_inst;

--> register_1_inst @0x0 , regisetr_2_inst @0x4

Is there a way to configurate the compiler to define a different offset for the registers? , for instance, the same example from above would lead to the following addresses:
--> register_1_inst @0x0 , regisetr_2_inst **@0x8**

Thanks

amykyta3, 2022-03-22:
You can use the alignment property to automatically infer additional padding between registers:
```
addrmap top {
    alignment = 8;
    register_1 register_1_inst;
    register_2 regisetr_2_inst;
};
```
This will ensure that, unless explicitly assigned, will be packed at addresses that are multiples of 8.

markcohen32, 2022-03-22:
Thanks, Also - is there a built-in way to verify that the registers do not violate the alignment in cases that their addresses are defined explicitly?

amykyta3, 2022-04-28:
Sorry for the late reply. I missed this followup question.

The current compiler doesn't have an explicit check against alignment currently - in my interpretation of the spec, the alignment property didn't seem like something that was intended to _enforce_ alignment, rather it only influences the alignment.

However, I do have some optional checkers you can enable as warnings or errors that may cover what you are looking for: https://systemrdl-compiler.readthedocs.io/en/stable/api/messages.html#warnings

Otherwise, I'd suggest adding your own post-elaboration step that further validates the design against your org-specific policies. Here is an example of one I added to the regblock project: https://github.com/SystemRDL/PeakRDL-regblock/blob/main/src/peakrdl/regblock/scan_design.py#L12

closed 2022-04-29 by amykyta3